openshwprojects/OpenBK7231T_App #1094: Missing expected MQTT messages?

TurkeyMan, 2024-02-18:
**Describe the bug**
According to this document: https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/mqttTopics.md
It says `rssi`, `ip`, `mac`, `datetime`, etc, etc are all "Send on connect and every N seconds (default: 60, if enabled)", I don't get this data on connect, and I can't see where to enable it... update the docs with instructions maybe?

**Firmware:**
- Version 1.17.464
- Chip/model: BK7231N
- Device: relay + energy meter
- Relay on channel 0 + BL0942

TurkeyMan, 2024-02-18:
I left it going for ages, and I did eventually see a couple of these messages, but I was expecting them on connect as described.

stefan064, 2024-02-26:
I had this too, check that Flag 2 is enabled _" [MQTT] Broadcast self state every N (def: 60) seconds (delay configurable by 'mqtt_broadcastInterval' and 'mqtt_broadcastItemsPerSec' commands)..."
You might need to be conservative with the publishing rate, ive seen devices have trouble with intermittent rebooting on high mqtt load and poor wifi. Increasing mqtt_broadcastInterval helped.