openshwprojects/OpenBK7231T_App #1100: [Feature request] New flag for Domoticz sensor data logging

vinibali, 2024-02-23:
**Describe the bug**
Hi! I'm using the Tasmota discovery plugin in Domoticz for all my IoT devices, it's a realible MQTT based discovery plugin.
Unfortunately I couldn't find the needed flags to make the OpenBK MQTT messages to be used with that environment.

**Firmware:**
- Version 1.17.424
- Device [CBU with CH8310](https://www.elektroda.com/rtvforum/topic4027957.html)
- Chip/model: BK7231N
- Device config NA

**To Reproduce**
Steps to reproduce the behavior:
1. Enable the MQTT in device
2. Let the sensor send the data to MQTT server
3. Device get's not identified by Tasmota Discovery plugin

Domoticz has HA plugin, but that's is not so configurabel. Could we do something in this topic?
How i see the MQTT messages are somewhat different, as Tasmota is sending the sensor data in JSON format, like this:
```
tele/tasmota_28/STATE {"Time":"2024-02-20T15:48:48","Uptime":"27T00:20:09","UptimeSec":2334009,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":4,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"MainWifi","BSSId":"FF:FF:FF:FF:FF:FF","Channel":6,"Mode":"11n","RSSI":100,"Signal":-36,"LinkCount":4,"Downtime":"0T00:01:03"}}
tele/tasmota_28/SENSOR {"Time":"2024-02-20T15:48:48","ANALOG":{"A0":3},"DS18B20":{"Id":"FF","Temperature":18.6},"TempUnit":"C"}
and OpenBK sends in this format (but you probably knows this :) )
```

tele/tasmota_obkE192184F/STATE {"Time":"1970-01-01T00:00:00","Uptime":"0T00:06:49","UptimeSec":409,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":23,"Vcc":2.4219,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MainWifi","BSSId":"FF:FF:FF:FF:FF:FF","Channel":11,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":21,"Downtime":"0T06:13:34"}}
tasmota_obkE192184F/1/get 0
tele/tasmota_obkE192184F/STATE {"Time":"1970-01-01T00:00:00","Uptime":"0T00:06:49","UptimeSec":409,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":23,"Vcc":2.4219,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MainWifi","BSSId":"FF:FF:FF:FF:FF:FF","Channel":11,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":21,"Downtime":"0T06:13:34"}}
tasmota_obkE192184F/2/get 290
tele/tasmota_obkE192184F/STATE {"Time":"1970-01-01T00:00:00","Uptime":"0T00:06:49","UptimeSec":409,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":23,"Vcc":2.4219,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MainWifi","BSSId":"FF:FF:FF:FF:FF:FF","Channel":11,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":21,"Downtime":"0T06:13:34"}}
tasmota_obkE192184F/3/get 62
```
Stream 1 is the state of the LED, 2 is the temperature without the decimal separator and 3 is the humidity.
And according to that the temperature is a 10 times higher value in the charts.
What is the reason for the battery voltage to be sent in the STATUS message?

Thanks

openshwprojects, 2024-02-27:
Sorry for the late reply. The formatting makes this issue very hard to read.

I'd love to add Domoticz support, but I need to know what exactly I need to send under which topics.

"And according to that the temperature is a 10 times higher value in the charts." - there is a flag for that, you can apply multiplier before mqtt publish

"What is the reason for the battery voltage to be sent in the STATUS message" - what do you mean, is Battery state missing for you?

Have you tried to enable Tasmota TELE publishes in flags?

vinibali, 2024-03-09:
sorry for the same from my side :)
* thanks for the multiplier flag, I've found it!
* yap, the "Flag 30 - [MQTT] Enable Tasmota TELE etc publishes (for ioBroker etc)" is enabled
_what do you mean, is Battery state missing for you?_
no, i have that that would be better in my case to have it in the sensor topic.
i've send you some Tasmota related MQTT message examples in the previous comment. It's seems the formatting is broken, I'll fix that.
So here is the Tasmota example:
```
tele/tasmota_28/STATE {"Time":"2024-02-20T15:48:48","Uptime":"27T00:20:09","UptimeSec":2334009,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":4,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"MainWifi","BSSId":"FF:FF:FF:FF:FF:FF","Channel":6,"Mode":"11n","RSSI":100,"Signal":-36,"LinkCount":4,"Downtime":"0T00:01:03"}}
tele/tasmota_28/SENSOR {"Time":"2024-02-20T15:48:48","ANALOG":{"A0":3},"DS18B20":{"Id":"FF","Temperature":18.6},"TempUnit":"C"}
```
BR

vinibali, 2024-04-21:
@openshwprojects did you have time to take a look?
Best regards

vinibali, 2024-09-29:
my use case have been solved with the following post from @Jasper2991:
https://github.com/openshwprojects/OpenBK7231T_App/issues/391#issuecomment-1301176765

closed 2024-09-29 by vinibali