openshwprojects/OpenBK7231T_App #1101: Publish some software identifying material?

TurkeyMan, 2024-02-24:
I want to detect/determine what software and version devices are running where possible.
There's quite a lot of device info/config that the device can report already; ie, MAC, IP, SSID, wifi channel, uptime, etc, etc.
Please also publish the software (ie, "OpenBeken"), and the version number among this device info?

TurkeyMan, 2024-02-24:
Also the chipset would be very valuable info to report for management UI and such.
An auto-update implementation needs to know this stuff.

TurkeyMan, 2024-02-24:
I just realised that `build` has the version, not just the build date. But the actual software name, and chipset would be good.

divadiow, 2024-02-24:
Would LOVE to see full flash ID, size and current use/free levels. Also the module and TuyaMCU baud, if present, in the main GUI

openshwprojects, 2024-02-24:
So you want to modify this str>
[image]

TurkeyMan, 2024-02-25:
I don't know if modify that string or add a new topic; your call on that!
Maybe a string like:
`"OpenBeken " USER_SW_VER " build: "__DATE__ " " __TIME__ " " hw: " [ie, BK7231N]`
Responding to @divadiow, maybe flash id can fit in there too? I reckon use/free levels and things like MCU baud rate might be different topic's though?

divadiow, 2024-02-25:
Ok sure. I just like seeing all the infos. And it would be handy, especially in the case of BL602, to know if the full chip info is something like BL602C20 or C10, so people know if it's a 1 or 2mb flash when reporting on potential firmware upgrade flash size issues.

stefan064, 2024-02-26:
Might be of some help, the home assistant discovery topic includes build verision and build platform e.g. `/homeassistant/sensor[light,swtich...]/device_full_name/config`, as Json ,e.g.:
 ```
{
  "dev": {
    "ids": [
      "OpenBK7231N_power_monitoring_plug"
    ],
    "name": "Aircon Bedroom",
    "sw": "1.17.483",                   <--------
    "mf": "Beken Corporation",
    "mdl": "BK7231N",                   <--------
    "cu": "http://192.168.0.102/index"
  },
```

TurkeyMan, 2024-02-27:
Interesting. Thanks for the tip; although probably not really addressing this issue. I don't think we should have to rely on homeassistant discovery (which is only triggered on demand) to obtain these values.
That manufacturer string is good to know about though; I would like to see that also added to the topic that I described in the OP here.
Do devices also have a hardware embedded model number or device identifier?

openshwprojects, 2024-02-27:
Would this change solve the issue:
https://github.com/openshwprojects/OpenBK7231T_App/pull/1107

openshwprojects, 2024-04-05:
Ok, I think it's safe to mark it as solved and close the issue.

closed 2024-04-05 by openshwprojects