openshwprojects/OpenBK7231T_App #1106: Remove "connected" topic?

hojnikb, 2024-02-26:
Is there a way to remove this topic from being published to the HA?

I'm having issues with my battery powered devices going "unavailable" after some times and i think this topic is the culprint.

Thank you!

openshwprojects, 2024-02-27:
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/flags.md

<html>
<body>

35 | [HASS] Deactivate avty_t flag for sensor when publishing to HASS (permit to keep value). You must restart HASS discovery for change to take effect.
-- | --

</body>
</html>

stefan064, 2024-02-28:
I separately just encountered this - Perhaps it may help to change the description for that option to something along the lines of "Deactivate sensor availability to prevent HASS from reporting 'unavailable' on device disconnect. Useful for battery powered devices[....??....]'