openshwprojects/OpenBK7231T_App #1108: Keeps restarting

Magnus-rosenborg, 2024-03-02:
**Describe the bug**
The firmware boots up normal and after a minute or so it restarts again. And so it keeps on doing. It connects to the mqtt server and while its up all works fine then restarts.. 

**Firmware:**
- Build: Build on Mar 1 2024 16:43:22 version 1.17.490
- mini smart switch
- Chip/model: BK7231N
- in autoexec.bat

startDriver ntp
ntp_setServer 193.11.166.8
ntp_setLatlong 56.017833, 14.268284
PowerSave 0
AddChangeHandler Channel2 < 1 publish Bekens91/KNAPPS1S2 TOGGLE -1
AddChangeHandler Channel2 > 0 publish Bekens91/KNAPPS1S2 TOGGLE -1
____________________-
Flags are: 1075840128

**To Reproduce**
I have tried with and without autoexec.bat filled in no differense.
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
I have 4 pieces of those. only tested on 2 but they behave the same.

Magnus-rosenborg, 2024-03-02:
Here is the log with extra debug:
from booting up until it restarts:

ExtraDebug:HTTP:Received 461 bytes from 192.168.200.254
ExtraDebug:HTTP:data: NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=120
LOCATION: http://192.168.200.254:5000/rootDesc.xml
SERVER: Linux/5.4.0-122-generic UPnP/1.1 MiniUPnPd/1.9
NT: urn:schemas-upnp-org:device:WANConnectionDevice:2
USN: uuid:ca5f9e2b-156b-48d1-8b3f-37d6bc370058::urn:schemas-upnp-org:device:WANConnectionDevice:2
NTS: ssdp:alive
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
01-NLS: 1
BOOTID.UPNP.ORG: 1
CONFIGID.UPNP.ORG: 1337

ExtraDebug:HTTP:Received 455 bytes from 192.168.200.254
ExtraDebug:HTTP:data: NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=120
LOCATION: http://192.168.200.254:5000/rootDesc.xml
SERVER: Linux/5.4.0-122-generic UPnP/1.1 MiniUPnPd/1.9
NT: urn:schemas-upnp-org:service:WANIPConnection:2
USN: uuid:ca5f9e2b-156b-48d1-8b3f-37d6bc370058::urn:schemas-upnp-org:service:WANIPConnection:2
NTS: ssdp:alive
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
01-NLS: 1
BOOTID.UPNP.ORG: 1
CONFIGID.UPNP.ORG: 1337

Info:MAIN:Time 23084, idle 167060/s, free 67992, MQTT 1(162), bWifi 1, secondsWithNoPing -1, socks 3/38 
ExtraDebug:HTTP:Received 457 bytes from 192.168.200.254
ExtraDebug:HTTP:data: NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=120
LOCATION: http://192.168.200.254:5000/rootDesc.xml
SERVER: Linux/5.4.0-122-generic UPnP/1.1 MiniUPnPd/1.9
NT: urn:schemas-upnp-org:service:Layer3Forwarding:1
USN: uuid:ca5f9e2b-156b-48d1-8b3f-37d6bc370056::urn:schemas-upnp-org:service:Layer3Forwarding:1
NTS: ssdp:alive
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
01-NLS: 1
BOOTID.UPNP.ORG: 1
CONFIGID.UPNP.ORG: 1337

ExtraDebug:HTTP:Received 473 bytes from 192.168.200.254
ExtraDebug:HTTP:data: NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=120
LOCATION: http://192.168.200.254:5000/rootDesc.xml
SERVER: Linux/5.4.0-122-generic UPnP/1.1 MiniUPnPd/1.9
NT: urn:schemas-upnp-org:service:WANCommonInterfaceConfig:1
USN: uuid:ca5f9e2b-156b-48d1-8b3f-37d6bc370057::urn:schemas-upnp-org:service:WANCommonInterfaceConfig:1
NTS: ssdp:alive
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
01-NLS: 1
BOOTID.UPNP.ORG: 1
CONFIGID.UPNP.ORG: 1337

ExtraDebug:HTTP:Received 457 bytes from 192.168.200.254
ExtraDebug:HTTP:data: NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=120
LOCATION: http://192.168.200.254:5000/rootDesc.xml
SERVER: Linux/5.4.0-122-generic UPnP/1.1 MiniUPnPd/1.9
NT: urn:schemas-upnp-org:service:DeviceProtection:1
USN: uuid:ca5f9e2b-156b-48d1-8b3f-37d6bc370056::urn:schemas-upnp-org:service:DeviceProtection:1
NTS: ssdp:alive
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
[... 300 lines cut ...]
NOTIFY * HTTP/1.1
SERVER: OpenBk
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.200.236:80/ssdp.xml
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:068462cc-c050-3daa-397d-3fc2027991b2::upnp:rootdevice

ExtraDebug:HTTP:DRV_SSDP_Send_Notify: sent message 245 bytes
Info:MAIN:Time 23137, idle 187240/s, free 67408, MQTT 1(162), bWifi 1, secondsWithNoPing -1, socks 4/38 
ExtraDebug:HTTP:Received 245 bytes from 192.168.200.236
ExtraDebug:HTTP:data: NOTIFY * HTTP/1.1
SERVER: OpenBk
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.200.236:80/ssdp.xml
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:068462cc-c050-3daa-397d-3fc2027991b2::upnp:rootdevice

ExtraDebug:HTTP:NOTIFY from a peer device
ExtraDebug:HTTP:SSDP obk device still present 0xecc8a8c0
Info:MAIN:Time 23138, idle 177304/s, free 67616, MQTT 1(162), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MAIN:Time 23139, idle 191153/s, free 67616, MQTT 1(162), bWifi 1, secondsWithNoPing -1, socks 3/38

openshwprojects, 2024-03-03:
Does it also happen with SSDP disabled?

Magnus-rosenborg, 2024-03-03:
I have not started it by myself. How do i disable it?

Magnus-rosenborg, 2024-03-03:
I have tried to pressed stop all drivers button on the tools

The log show this i it keeps on restarting.
Here comes the log.

 wifi changes
Info:MAIN:Time 64582, idle 180713/s, free 70096, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64583, idle 189339/s, free 68672, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64584, idle 188834/s, free 68400, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64585, idle 89579/s, free 68400, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64586, idle 0/s, free 68400, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64587, idle 0/s, free 68400, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTING - 1
Info:MAIN:Time 64588, idle 98988/s, free 68976, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64589, idle 190286/s, free 68976, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64590, idle 191212/s, free 68976, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:GEN:dhcp=0 ip=0.0.0.0 gate=0.0.0.0 mask=0.0.0.0 mac=c8:47:8c:00:00:00
Info:GEN:sta: 0, softap: 0, b/g/n
Info:MAIN:Time 64591, idle 189591/s, free 68976, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64592, idle 191275/s, free 68976, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64593, idle 190659/s, free 68976, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64594, idle 191399/s, free 68976, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64595, idle 190259/s, free 68976, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64596, idle 190382/s, free 68976, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64597, idle 191561/s, free 68976, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_AUTH_FAILED - 3
Info:MAIN:Time 64598, idle 185684/s, free 68288, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64599, idle 189210/s, free 67744, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64600, idle 190418/s, free 67400, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:GEN:dhcp=0 ip=0.0.0.0 gate=0.0.0.0 mask=0.0.0.0 mac=c8:47:8c:00:00:00
Info:GEN:sta: 0, softap: 0, b/g/n
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTING - 1
Info:MAIN:Time 64601, idle 181321/s, free 68896, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64602, idle 191155/s, free 68896, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64603, idle 191244/s, free 68896, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64604, idle 190109/s, free 68896, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64605, idle 191380/s, free 68896, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64606, idle 190964/s, free 68896, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64607, idle 189993/s, free 68896, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64608, idle 190343/s, free 68896, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64609, idle 190924/s, free 68896, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64610, idle 190360/s, free 68896, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:GEN:dhcp=0 ip=0.0.0.0 gate=0.0.0.0 mask=0.0.0.0 mac=c8:47:8c:00:00:00
Info:GEN:sta: 0, softap: 0, b/g/n
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_AUTH_FAILED - 3
Info:MAIN:Time 64611, idle 187097/s, free 68216, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64612, idle 190180/s, free 67672, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTING - 1
Info:MAIN:Time 64613, idle 374214/s, free 68744, MQTT 0(439), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED - 4
Info:MAIN:Time 64614, idle 175076/s, free 68992, MQTT 0(439), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64615, idle 206833/s, free 69032, MQTT 0(439), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MQTT:mqtt_userName scriptuser
mqtt_pass 123qaz
mqtt_clientID Bekens91
mqtt_host 192.168.200.1:1883
Info:MAIN:Time 64616, idle 193951/s, free 60336, MQTT 0(440), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MQTT:mqtt_connection_cb: Successfully connected
Info:MQTT:mqtt_subscribed to Bekens91/+/set
Info:MQTT:mqtt_subscribed to bekens/+/set
Info:MQTT:mqtt_subscribed to cmnd/Bekens91/+
Info:MQTT:mqtt_subscribed to cmnd/bekens/+
Info:MQTT:mqtt_subscribed to Bekens91/+/get
Info:MQTT:mqtt_subscribed to tele/Bekens91/+
Info:MQTT:mqtt_subscribed to stat/Bekens91/+
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/Bekens91/POWER48
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/Bekens91/POWER
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic Bekens91/48/get
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic Bekens91/49/get
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic Bekens91/KNAPPS1S2/get
[... 134 lines cut ...]
Info:MAIN:Time 64666, idle 195720/s, free 68944, MQTT 1(440), bWifi 1, secondsWithNoPing -1, socks 2/38 
[... cut ...]
Info:GEN:sta:rssi=-69,ssid=lj3m,bssid=88:3d:24:c9:eb:10,channel=6,cipher_type:CCMP
Info:MAIN:Time 64681, idle 189067/s, free 68944, MQTT 1(440), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64682, idle 195057/s, free 68944, MQTT 1(440), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64683, idle 190053/s, free 68944, MQTT 1(440), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 64684, idle 191005/s, free 68944, MQTT 1(440), bWifi 1, secondsWithNoPing -1, socks 2/38

openshwprojects, 2024-03-03:
So both autoexec.bat and startup commands are empty?

Magnus-rosenborg, 2024-03-03:
[image]
No autoexec.bat or any other file.
But still restarting

openshwprojects, 2024-03-04:
In the Web App log, does the free memory amount goes down? It doesnt look like so.

Is device brand new?

Can you add PowerSave 1 to startup command and reboot?

You may have device with this fault: https://www.elektroda.pl/rtvforum/topic3898805.html

Magnus-rosenborg, 2024-03-04:
About Memory i can check later.  Yes, brand New all of them. Med vänlig hälsning, Magnus

Magnus-rosenborg, 2024-03-05:
Youre right the memory is stable, no difference with powersave =1 or 0. My unit is quiet also.

JohnyPeaN, 2024-06-04:
Hi, is there some progress in finding the culprit?
I realized that all my devices restart. Didn't notice it on battery powered sleeping devices, but all powerplugs and now light switch, too.
I'll add some observations here:

Reboot reason is always: 1 - Rbt
All are BK7231N devices.

Now that the firmware reports temp and uptime, I can monitor it:
It seems random, but reboots are between 2000 to 40000 seconds apart.
Reported temperature is around 36 with 42 max. I don't suspect powersource, because I have replaced one module with esp tasmota even with aditional temp/hum sensor and its completely stable (> month uptime).

While comparing plug that powers freezer, I see temperature rising to 40 while freezer unit active and dropping to 37 while inactive. But reboots don't really follow a pattern. This one has them more frequent overall. Period 3000s. RSI -62.

Plug powering small led lamp has higher temp = 42 that doesn't fluctuate much. It has either 0W or ~2W. Reboots are less frequent. Period avg 26000s. RSI -6 (very near AP).

Another one with stable consumption ~20W has reboot period on avg 17000s. RSI -45.

Wall light switch has temperature 36, period ~16000s and RSI -47.

Signal strength seems to be only thing that remotely seems to be affecting the reboot frequency. But its a far stretch and just a few always online devices. Also plugs with nothing connected to them reboot.
Powersave 0/1 doesn't seem to have an effect.
Everything is connected to Homeassistant and except the reboots works flawlessly.

If there is something I could test. Let me know.

Graphs for the cooler connected plug:
[image]
[image]
[image]

JohnyPeaN, 2024-09-02:
Haven't tracked it down to speciffic version, but after update from ~600 to 640 (version 1.17.640 and later work perfectly) the problem was solved on all devices. Uptime reaching weeks.