openshwprojects/OpenBK7231T_App #1112: support for BK7231M(C2BS)

shzlww, 2024-03-05:
**Describe the bug**
I have a smart plug(not Tuya version) with cb2s module ,but the module is using a BK7231M. When selecting BK7231N, the Module would flash just fine, but when power cycling it, it would not boot and the AP would not come up.

**Screenshots**
[image]

**Additional context**
Original firmware dump by bk7231flasher_1.1.6
[readResult_BK7231N_QIO_2024-02-3-18-21-04.bin.gz](https://github.com/openshwprojects/OpenBK7231T_App/files/14491950/readResult_BK7231N_QIO_2024-02-3-18-21-04.bin.gz)

backup and write logs:

> Backup name has not been set, so output file will only contain flash type/date.
> 
> Starting read!
> Now is: 2024年3月2日 18:17:31.
> Flasher mode: BK7231N
> Going to open port: COM3.
> Serial port open!
> Getting bus... (now, please do reboot by CEN or by power off/on)
> Getting bus failed, will try again - 0/100!
> Getting bus success!
> Going to set baud rate setting (921600)!
> Failed to set baud rate!
> There was no result to save.
> Backup name has not been set, so output file will only contain flash type/date.
> Starting read!
> Now is: 2024年3月2日 18:17:50.
> Flasher mode: BK7231N
> Going to open port: COM3.
> Serial port open!
> Getting bus... (now, please do reboot by CEN or by power off/on)
> Getting bus success!
> Going to set baud rate setting (115200)!
> Will try to read device flash MID (for unprotect N):
> Flash MID loaded: 1560EB
> Will now search for Flash def in out database...
> Flash def found! For: 1560EB
> Flash information: mid: 1560EB, icName: TH25Q_16HB, manufacturer: TH, szMem: 1000000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-FF-FF-FF
> Entering SetProtectState(True)...
> sr: 4
> sr: 4004
> final sr: 4004
> msk: 407c
> cw: 0, sb: 2, lb: 5
> bfd: 0
> sr: 0
> sr: 0
> final sr: 0
> msk: 407c
> cw: 0, sb: 2, lb: 5
> bfd: 0
> SetProtectState(True) success!
> Going to start reading at offset 0x00...
> Reading 0x00... Ok! Reading 0x1000... Ok! Reading 0x2000... Ok! Reading 0x3000... Ok! Reading 0x4000... Ok! Reading 0x5000... Ok! Reading 0x6000... Ok! Reading 0x7000... Ok! Reading 0x8000... Ok! Reading 0x9000... Ok! Reading 0xA000... Ok! Reading 0xB000... Ok! Reading 0xC000... Ok! Reading 0xD000... Ok! Reading 0xE000... Ok! Reading 0xF000... Ok! Reading 0x10000... Ok! Reading 0x11000... Ok! Reading 0x12000... Ok! Reading 0x13000... Ok! Reading 0x14000... Ok! Reading 0x15000... Ok! Reading 0x16000... Ok! Reading 0x17000... Ok! Reading 0x18000... Ok! Reading 0x19000... Ok! Reading 0x1A000... Ok! Reading 0x1B000... Ok! Reading 0x1C000... Ok! Reading 0x1D000... Ok! Reading 0x1E000... Ok! Reading 0x1F000... Ok! Reading 0x20000... Ok! Reading 0x21000... Ok! Reading 0x22000... Ok! Reading 0x23000... Ok! Reading 0x24000... Ok! Reading 0x25000... Ok! Reading 0x26000... Ok! Reading 0x27000... Ok! Reading 0x28000... Ok! Reading 0x29000... Ok! Reading 0x2A000... Ok! Reading 0x2B000... Ok! Reading 0x2C000... Ok! Reading 0x2D000... Ok! Reading 0x2E000... Ok! Reading 0x2F000... Ok! Reading 0x30000... Ok! Reading 0x31000... Ok! Reading 0x32000... Ok! Reading 0x33000... Ok! Reading 0x34000... Ok! Reading 0x35000... Ok! Reading 0x36000... Ok! Reading 0x37000... Ok! Reading 0x38000... Ok! Reading 0x39000... Ok! Reading 0x3A000... Ok! Reading 0x3B000... Ok! Reading 0x3C000... Ok! Reading 0x3D000... Ok! Reading 0x3E000... Ok! Reading 0x3F000... Ok! Reading 0x40000... Ok! Reading 0x41000... Ok! Reading 0x42000... Ok! Reading 0x43000... Ok! Reading 0x44000... Ok! Reading 0x45000... Ok! Reading 0x46000... Ok! Reading 0x47000... Ok! Reading 0x48000... Ok! Reading 0x49000... Ok! Reading 0x4A000... Ok! Reading 0x4B000... Ok! Reading 0x4C000... Ok! Reading 0x4D000... Ok! Reading 0x4E000... Ok! Reading 0x4F000... Ok! Reading 0x50000... Ok! Reading 0x51000... Ok! Reading 0x52000... Ok! Reading 0x53000... Ok! Reading 0x54000... Ok! Reading 0x55000... Ok! Reading 0x56000... Ok! Reading 0x57000... Ok! Reading 0x58000... Ok! Reading 0x59000... Ok! Reading 0x5A000... Ok! Reading 0x5B000... Ok! Reading 0x5C000... Ok! Reading 0x5D000... Ok! Reading 0x5E000... Ok! Reading 0x5F000... Ok! Reading 0x60000... Ok! Reading 0x61000... Ok! Reading 0x62000... Ok! Reading 0x63000... Ok! Reading 0x64000... Ok! Reading 0x65000... Ok! Reading 0x66000... Ok! Reading 0x67000... Ok! Reading 0x68000... Ok! Reading 0x69000... Ok! Reading 0x6A000... Ok! Reading 0x6B000... Ok! Reading 0x6C000... Ok! Reading 0x6D000... Ok! Reading 0x6E000... Ok! Reading 0x6F000... Ok! Reading 0x70000... Ok! Reading 0x71000... Ok! Reading 0x72000... Ok! Reading 0x73000... Ok! Reading 0x74000... Ok! Reading 0x75000... Ok! Reading 0x76000... Ok! Reading 0x77000... Ok! Reading 0x78000... Ok! Reading 0x79000... Ok! Reading 0x7A000... Ok! Reading 0x7B000... Ok! Reading 0x7C000... Ok! Reading 0x7D000... Ok! Reading 0x7E000... Ok! Reading 0x7F000... Ok! Reading 0x80000... Ok! Reading 0x81000... Ok! Reading 0x82000... Ok! Reading 0x83000... Ok! Reading 0x84000... Ok! Reading 0x85000... Ok! Reading 0x86000... Ok! Reading 0x87000... Ok! Reading 0x88000... Ok! Reading 0x89000... Ok! Reading 0x8A000... Ok! Reading 0x8B000... Ok! Reading 0x8C000... Ok! Reading 0x8D000... Ok! Reading 0x8E000... Ok! Reading 0x8F000... Ok! Reading 0x90000... Ok! Reading 0x91000... Ok! Reading 0x92000... Ok! Reading 0x93000... Ok! Reading 0x94000... Ok! Reading 0x95000... Ok! Reading 0x96000... Ok! Reading 0x97000... Ok! Reading 0x98000... Ok! Reading 0x99000... Ok! Reading 0x9A000... Ok! Reading 0x9B000... Ok! Reading 0x9C000... Ok! Reading 0x9D000... Ok! Reading 0x9E000... Ok! Reading 0x9F000... Ok! Reading 0xA0000... Ok! Reading 0xA1000... Ok! Reading 0xA2000... Ok! Reading 0xA3000... Ok! Reading 0xA4000... Ok! Reading 0xA5000... Ok! Reading 0xA6000... Ok! Reading 0xA7000... Ok! Reading 0xA8000... Ok! Reading 0xA9000... Ok! Reading 0xAA000... Ok! Reading 0xAB000... Ok! Reading 0xAC000... Ok! Reading 0xAD000... Ok! Reading 0xAE000... Ok! Reading 0xAF000... Ok! Reading 0xB0000... Ok! Reading 0xB1000... Ok! Reading 0xB2000... Ok! Reading 0xB3000... Ok! Reading 0xB4000... Ok! Reading 0xB5000... Ok! Reading 0xB6000... Ok! Reading 0xB7000... Ok! Reading 0xB8000... Ok! Reading 0xB9000... Ok! Reading 0xBA000... Ok! Reading 0xBB000... Ok! Reading 0xBC000... Ok! Reading 0xBD000... Ok! Reading 0xBE000... Ok! Reading 0xBF000... Ok! Reading 0xC0000... Ok! Reading 0xC1000... Ok! Reading 0xC2000... Ok! Reading 0xC3000... Ok! Reading 0xC4000... Ok! Reading 0xC5000... Ok! Reading 0xC6000... Ok! Reading 0xC7000... Ok! Reading 0xC8000... Ok! Reading 0xC9000... Ok! Reading 0xCA000... Ok! Reading 0xCB000... Ok! Reading 0xCC000... Ok! Reading 0xCD000... Ok! Reading 0xCE000... Ok! Reading 0xCF000... Ok! Reading 0xD0000... Ok! Reading 0xD1000... Ok! Reading 0xD2000... Ok! Reading 0xD3000... Ok! Reading 0xD4000... Ok! Reading 0xD5000... Ok! Reading 0xD6000... Ok! Reading 0xD7000... Ok! Reading 0xD8000... Ok! Reading 0xD9000... Ok! Reading 0xDA000... Ok! Reading 0xDB000... Ok! Reading 0xDC000... Ok! Reading 0xDD000... Ok! Reading 0xDE000... Ok! Reading 0xDF000... Ok! Reading 0xE0000... Ok! Reading 0xE1000... Ok! Reading 0xE2000... Ok! Reading 0xE3000... Ok! Reading 0xE4000... Ok! Reading 0xE5000... Ok! Reading 0xE6000... Ok! Reading 0xE7000... Ok! Reading 0xE8000... Ok! Reading 0xE9000... Ok! Reading 0xEA000... Ok! Reading 0xEB000... Ok! Reading 0xEC000... Ok! Reading 0xED000... Ok! Reading 0xEE000... Ok! Reading 0xEF000... Ok! Reading 0xF0000... Ok! Reading 0xF1000... Ok! Reading 0xF2000... Ok! Reading 0xF3000... Ok! Reading 0xF4000... Ok! Reading 0xF5000... Ok! Reading 0xF6000... Ok! Reading 0xF7000... Ok! Reading 0xF8000... Ok! Reading 0xF9000... Ok! Reading 0xFA000... Ok! Reading 0xFB000... Ok! Reading 0xFC000... Ok! Reading 0xFD000... Ok! Reading 0xFE000... Ok! Reading 0xFF000... Ok! Reading 0x100000... Ok! Reading 0x101000... Ok! Reading 0x102000... Ok! Reading 0x103000... Ok! Reading 0x104000... Ok! Reading 0x105000... Ok! Reading 0x106000... Ok! Reading 0x107000... Ok! Reading 0x108000... Ok! Reading 0x109000... Ok! Reading 0x10A000... Ok! Reading 0x10B000... Ok! Reading 0x10C000... Ok! Reading 0x10D000... Ok! Reading 0x10E000... Ok! Reading 0x10F000... Ok! Reading 0x110000... Ok! Reading 0x111000... Ok! Reading 0x112000... Ok! Reading 0x113000... Ok! Reading 0x114000... Ok! Reading 0x115000... Ok! Reading 0x116000... Ok! Reading 0x117000... Ok! Reading 0x118000... Ok! Reading 0x119000... Ok! Reading 0x11A000... Ok! Reading 0x11B000... Ok! Reading 0x11C000... Ok! Reading 0x11D0 [... cut ...] ector 0x120000... ok! Writing sector 0x121000... ok! Writing sector 0x122000... ok! Writing sector 0x123000... ok! Writing sector 0x124000... ok! Writing sector 0x125000... ok! Writing sector 0x126000... ok! Writing sector 0x127000... ok! Writing sector 0x128000... ok! Writing sector 0x129000... ok! Starting CRC check for 298 sectors, starting at offset 00
> CRC matches 0x5A99C1C1!
> 
> NOTE: the OBK config writing is disabled, so not writing anything extra.
> Writing file data to chip successs.

shzlww, 2024-03-06:
@openshwprojects Any suggestion?

xjzsq, 2024-03-06:
Same here.
It seems I have the same smart plug as you (Hejiaqin/Huiju X1S), the component solution is CB2S with bk7231n + BL0937 + YX201T-S-105DM. Looking forward to a solution.
btw, this is the original firmware for this plug. I hope it can be helpful in resolving the issue.
[readResult_BK7231N_QIO_1_2024-06-3-22-03-37.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/14513957/readResult_BK7231N_QIO_1_2024-06-3-22-03-37.zip)

openshwprojects, 2024-03-08:
I have one M chip. I need to find some time to check this out, but there are also many other feature requests. How many of people reading has this issue? @shzlww  and @xjzsq ?

Beat0small0Han, 2024-03-08:
same here

> I have one M chip. I need to find some time to check this out, but there are also many other feature requests. How many of people reading has this issue? @shzlww and @xjzsq ?

shzlww, 2024-03-09:
> I have one M chip. I need to find some time to check this out, but there are also many other feature requests. How many of people reading has this issue? @shzlww and @xjzsq ?

@openshwprojects  There are a lot of sockets using bk7231m in China, and the price is very cheap,about 2 dollars each. So, it would be wonderful if bk7231m  be supported by Openbeken. 
BTW I am also Seeking support from libretiny team:https://github.com/libretiny-eu/libretiny/issues/262

xjzsq, 2024-03-09:
> > I have one M chip. I need to find some time to check this out, but there are also many other feature requests. How many of people reading has this issue? @shzlww and @xjzsq ?
> 
> @openshwprojects There are a lot of sockets using bk7231m in China, and the price is very cheap,about 2 dollars each. So, it would be wonderful if bk7231m be supported by Openbeken. BTW I am also Seeking support from libretiny team:[libretiny-eu/libretiny#262](https://github.com/libretiny-eu/libretiny/issues/262)

Until this issue is resolved, I have discovered a method of intercepting and even tampering with the reporting information of the socket by means of a man-in-the-middle attack in order to integrate the socket with Home Assistant. However, I have not yet found the MQTT command for querying status. I would like to discuss this together and I would be grateful if you could reach out to me via email at [x@xjzsq.ren](mailto:x@xjzsq.ren). Looking forward to your response and also to the resolution of this issue.

xjzsq, 2024-03-09:
> @openshwprojects There are a lot of sockets using bk7231m in China, and the price is very cheap,about 2 dollars each. So, it would be wonderful if bk7231m be supported by Openbeken. BTW I am also [...]

Until this issue is resolved, I have discovered a method of intercepting and even tampering with the reporting information of the plug  in order to integrate the plug with Home Assistant. However, I have not yet found the MQTT command for querying status. I would like to discuss this together and I would be grateful if you could reach out to me via email at [x@xjzsq.ren](mailto:x@xjzsq.ren). Looking forward to your response and also to the resolution of this issue.

shzlww, 2024-03-10:
@xjzsq  https://github.com/libretiny-eu/libretiny/issues/262 Try this,I made a bold attempt,modified the binary code block of the firmware(may be bootloader,but i'm not sure),esphome seems to be working here,but need more test.