openshwprojects/OpenBK7231T_App #1113: Accidentally flashed OTA over CCtr, only see A-CBBA AP without any functionality

ashirviskas, 2024-03-05:
My first time flashing OpenBK, made this mistake, but flashing was successful.

Now I can get the device to produce an AP that is named `A-CBBA` and I can connect to it, but I get the `192.168.175.100/24` or `192.168.175.101/24` IPs and I cannot access the web UI. I tried `192.168.175.1` and as per some documentation `192.168.4.1` (tried manual IP config too), on my linux PC and on my android phone, but I got no response.

Any ideas?

openshwprojects, 2024-03-05:
OBK will not create AP like A-CBBA. The flashing must have failed.

ashirviskas, 2024-03-05:
Huh, does this AP name seem familiar? Maybe I accidentally flashed something else (esphome?), but search engines are useless for this name.

openshwprojects, 2024-03-05:
I would say it's Tuya AP pairing mode. The flashing has failed. Please retry floashing OBK and I think it will work (unless your device is patched). I can help with OBK configuration once it's working.

openshwprojects, 2024-03-09:
@ashirviskas hey, have you tried again, what's the state of this issue?

ashirviskas, 2024-03-11:
Hey, I had a busy week, so only now I got back to it. 

I did not have any success using the CCtr, so I soldered to serial, backed up the flash using `bk7231tools` (because `BK7231Flasher` did not work, probably due to the failed flash previously) and then installed the latest OpenBK. I got it to connect to the wifi and now I'm working on figuring out the configuration. Any tips?

Here is the picture of the Pir Motion Sensor disassembled (before my soldering attempt):

[image]

I'll make sure to share the config if/when I manage to get it functional.

ashirviskas, 2024-03-11:
Update: The configuration seems to be very similar to the one described [here](https://www.elektroda.com/rtvforum/topic3988602.html) (LED, Battery and Button pins are the same), but I did not manage to get the PIR sensor working properly.