openshwprojects/OpenBK7231T_App #1116: Unable to set static ip address

Magnus-rosenborg, 2024-03-07:
I have changed 0.0.0.0 to 192.168.200.239 but it's still using the old dhcp address.
The device have been swithed off but still the same problem.
Flags: 1120

**Firmware:**
- Version 1.17.494
- Device? mini smart switch
- Chip/model: BL602
- Device config? nothing. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
[image]
[image]
[image]

**Additional context**
Add any other context about the problem here.

Magnus-rosenborg, 2024-03-07:
Se also my reported issue about lfs not working on the same device.

openshwprojects, 2024-04-05:
This has been solved with PR #1155 . I am closing this issue now, but let me know if there is some other issues with static IP on BL602 platform.

closed 2024-04-05 by openshwprojects