openshwprojects/OpenBK7231T_App #1117: lfs not working

Magnus-rosenborg, 2024-03-07:
**Describe the bug**
I have tried to create autoexec.bat but it will only show me " Will try to create autoexec.bat. But nothing happens. When i try to list files nothing happens. i'm not verry familiary with Bekens cmd but also tried to format the lfs using telnet on port 100. see picture. I have 3 identical brand new peaces but yet only flashed 2. Both have the same issue (same version of firmware) I have also reported another bug on this, its about static ip not working.
Flags: 1120

**Firmware:**
- Version 1.17.494
- Device? mini smart switch
- Chip/model: BL602
- Device config? none

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
[image]
[image]

**Additional context**
Add any other context about the problem here.

openshwprojects, 2024-03-07:
LittleFS is currently unimplemented on BL602, but we may try to change that soon. If you have basic C skills, you can consider helping with that.

Magnus-rosenborg, 2024-03-08:
Ahhh, didn’t know that. Sorry. No i don’t have c-skills. Med vänlig hälsning, Magnus