openshwprojects/OpenBK7231T_App #1124: addClockEvent: Time parameter interpreted as TimerSeconds

Tekagi01, 2024-03-12:
If you enter a time in addClockEvent and add a leading zero to get the HH:mm or HH:mm:ss format (e.g. 09:00) the time seems to be interpreted as a "TimerSeconds" parameter instead of a "Time" parameter.

**Firmware:**
- Version 1.17.501

**To Reproduce**

1. add a clock event with time in format HH:mm or HH:mm:ss containing a leading zero:
`addClockEvent 09:00:00 0xff 1 setChannel 1 1`

2. listClockEvents
what it returns:
`Info:CMD:Ev 1 - 0:0:9, days 0xff, cmd setChannel 1 1 `
What it should return:
`Info:CMD:Ev 1 - 9:0:0, days 0xff, cmd setChannel 1 1`

**Additional context**
The commands.md says to enter time in the HH:mm or HH:mm:ss format.
This does not work in all cases.
Either the docs should be updated and clearly state that we have to omit leading zeros, or the interpretation of the command should also consider leading zeros in a "Time" parameter which would be the better way imho.

openshwprojects, 2024-03-12:
Guys @rickbronson @ilengyel can you check this out? Otherwise I will investigate soon. Maybe it's bad sscanf?

ilengyel, 2024-03-14:
Yep, the intention was to handle the leading 0s, I can take a look at it this evening.

ilengyel, 2024-03-14:
Odd, I can reproduce this issue in my device but not in the selftest code. I will try to debug this on my device or try setting up the simulator over the weekend.

ilengyel, 2024-03-15:
Actually, my testing was bad, I decided to use hour "03:00:00" in the selftest not "09:00:00". The reason that the former works and the latter does not is the choice of [format](https://learn.microsoft.com/en-us/cpp/c-runtime-library/scanf-type-field-characters?view=msvc-170) specifier in: `sscanf(s, "%i:%i:%i", &hour, &minute, &second)` the use of %i allows for integers formatted as decimal (prefixed with nothing), hex (prefixed with 0x) or octal (prefixed with 0) which is why 07:00:00 would work but 08 and 09 would break the parsing.

`sscanf(s, "%d:%d:%d", &hour, &minute, &second)` would be the fix.

closed 2024-03-15 by openshwprojects