openshwprojects/OpenBK7231T_App #1125: Internal Température Chipset .

nadlio, 2024-03-12:
**Describe the bug**
Hello, how to retrieve the internal temperature in HA, MQTT ?

Thanks

**Firmware:**

- Chip/model: BK7231T or Chipset BK7231N
- Device : plug 

[image]

openshwprojects, 2024-03-12:
Sure, wait a moment, I will add a var for that and give you a ready script

openshwprojects, 2024-03-13:
```
// addRepeatingEvent [IntervalSeconds][RepeatsOr-1]
addRepeatingEvent 5 -1 publishFloat SelfTemperature $intTemp

```
https://obrazki.elektroda.pl/5857802700_1710276456.png

nadlio, 2024-03-13:
Perfect +++
Thanks

icovelev, 2024-03-28:
Could this be added as part of the next releases, so there’s no need to use the script on every device or new install?