openshwprojects/OpenBK7231T_App #1126: SSID name with quote

Kozoya, 2024-03-12:
not able to connect with a SSID containing a single quote. eg: "I don't like this wifi"
wifi number 2 is then not used

**Firmware:**
- Version 1.17.501
- Device LED controller
- Chip/model: BL602

openshwprojects, 2024-03-18:
I've did some basic self-tests on WIndows and it does not look like it's a HTTP page issue. The special characters parsing works just fine. I need to test for it on BK7231 and BL602 to determine whether is BL602-specific...

giedriuslt, 2024-03-25:
Works fine for me:
[image]

Kozoya, 2024-03-28:
I catch this logs if that can help.
I changed de PSK  :-)

[logs.txt](https://github.com/openshwprojects/OpenBK7231T_App/files/14792772/logs.txt)

giedriuslt, 2024-03-28:
Looking for that error:
`Association failure, Wlan802.11 status code = 17`
I find this explanation
`Association denied because AP is unable to handle additional associated stations`
Maybe it was actually issue with router?

onlysli, 2024-11-05:
Hello. 
I have similar issue but when the SSID has dots inside, eq: "www.example.com".
On the same router I have another SSID setup that doesn't have dots. With that SSD it works. So the router is not the issue here.
I've also set the same encryption for both SSIDs and only to the one without dots it connects.

Firmware:
    Version 1.17.765
    Device LED controller
    Chip/model: BL602