openshwprojects/OpenBK7231T_App #1135: Can't not connect to wifi

nazubi222, 2024-03-22:
**Describe the bug**
After i erase all flash via BK7231 GUI tool and flash the firmware. I set SSID and password to connect but it can't. I've tried change MAC and restore RF part but it sill not working. 

How can i fix this?

**Firmware:**
- Version 1.17.515
- Device: Tuya CBU 
- Chip/model: BK7231N

rafalolb, 2024-03-23:
Additionally, I have a problem with configure secound Wifi. Second wifi not working now. This function is crucial for me.

Can you implement it if you will correct this bug?

divadiow, 2024-03-24:
> After i erase all flash via BK7231 GUI tool and flash the firmware. I set SSID and password to connect but it can't. I've tried change MAC and restore RF part but it sill not working.
> 
> How can i fix this?`

Have you tried reflashling factory firmware then doing OBK again?

nazubi222, 2024-03-25:
> > After i erase all flash via BK7231 GUI tool and flash the firmware. I set SSID and password to connect but it can't. I've tried change MAC and restore RF part but it sill not working.
> > How can i fix this?`
> 
> Have you tried reflashling factory firmware then doing OBK again?

I tried and it works. Thanks you

closed 2024-03-25 by nazubi222