openshwprojects/OpenBK7231T_App #1142: TglChanOnTgl not reliable on device with AC Frequency Detection (like Moes MS-104B) - change state without input

paede81, 2024-03-25:
**Describe the bug**
If you use a device with AC Frequency Detection (like Moes MS-104B) then TglChanOnTgl is not reliable. It switches the device multiple time a day with toggle the input

Please add something similar then `SwitchDebounce 69` to the application (as with Tasmota)

I believe bug is related to: https://github.com/openshwprojects/OpenBK7231T_App/issues/753

**Firmware:**
- Version: 1.17.496
- Device: Moes MS-104B 
- Chip/model: BK7231N
- Device config:
```
  "pins": {
    "6": "Rel;2",
    "8": "Rel;1",
    "24": "TglChanOnTgl;2",
    "26": "TglChanOnTgl;1"
```

**To Reproduce**
Steps to reproduce the behavior:
1. Set input to TglChanOnTgl
2. Wait...
3. Device will change input state with toggling any switch which which is connected

**Additional context**
Tasmota identified the issue and created a workaround:

```
AC Frequency Detection Switch[~](https://tasmota.github.io/docs/Buttons-and-Switches/#ac-frequency-detection-switch)
Some devices, such as [BlitzWolf BW-SS5](https://templates.blakadder.com/blitzwolf_BW-SS5.html) or [Moes MS-104B](https://templates.blakadder.com/moes-MS-104B.html), use mains frequency detection on their switch inputs. Whenever the connected switch or button is pressed there are 50/60 Hz pulses on the switch input. Inside the switch there's a frequency detection circuit which is connected to a GPIO of the ESP8266 chip which counts those pulses. Prior to Tasmota 8.4 this kind of switching was handled using Counter sensors and scripting which is now simplified.

You can imagine this algorithm as a leaking bucket. Every pulse adds water to the bucket (little more than leaking out in a cycle), but the water is dripping countinously. If the bucket is full, we will treat the switch on. If there's no pulses, the bucket will be empty, and the we will turn off the switch. The size of the bucket is the debouncing time which controls the sensitivity of the algorithm. If the mains frequency is 50 Hz, a whole AC wave is 20 msec long (for 60 Hz it's about 17 msec; 1000 / frequency if we want the result in milliseconds). The exact frequency is not really important, because we add more water for every pulse.

After you have assigned a Switch<x> to the GPIO connected to the AC frequency detection circuit use the ['SwitchDebounce'](https://tasmota.github.io/docs/Commands/#switchdebounce) command to set the number of pulses required for the switch to be recognized as on or off. For example: SwitchDebounce 69 will turn the switch on after three pulses and turn it off after three missing ones (3 * 20 msec is 60 and the last digit must be 9 to activate the AC detection). You will probably have to experiment with the values depending on your AC frequency and the devices frequency detection implementation.

Once the feature is enabled you can use this switch as any regular switch!
```

Code snipped here:
https://github.com/arendst/Tasmota/commit/7e8c1368a805ca0a634cf9f2608117072336cfe4

openshwprojects, 2024-04-05:
I haven't seen such device yet. I may need to buy one with donation funds if they are sufficient, or is there anyone with such device willing to help with testing?

paede81, 2024-04-14:
Sorry for the late response
I'm happy to help with testing. I have different device in use with AC Frequency Detection
Currently I'm using this device: https://www.aliexpress.com/item/1005003425831720.html
and this: https://www.aliexpress.com/item/1005006339647483.html

paede81, 2024-04-14:
It seems like other people are affected by the same bug:
https://www.elektroda.com/rtvforum/viewtopic.php?p=21038547#21038547
https://www.elektroda.com/rtvforum/viewtopic.php?p=20947436#20947436
https://www.elektroda.com/rtvforum/viewtopic.php?p=20934967#20934967

paede81, 2024-05-05:
@openshwprojects : can you please flag this issues as a bug as this makes device with ac detection unusable

rt400, 2024-08-18:
Hi. The problem still here with the latest firmware...