openshwprojects/OpenBK7231T_App #1144: ln882HKI becomes unresponsive after 2 hours

geniusled, 2024-03-25:
**Describe the bug**
After running the device for 2 hours, it becomes unresponsive. the following can be observed:
- not connected to wifi anymore
- ofcourse, no more mqtt
- doesnt respond to button press anymore (relay,light)
- lights stay on last state

**Firmware:**
- Version OpenLN882H_1.17.516.bin (Build on Mar 23 2024 18:37:28 version LN882H_Test)
- Device tuya power monitor socket (BSD01)
- Chip/model: Lightning LN882HKI
- Device config (copied from web application / config)

`{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Full Device Name Here",
  "model": "enter short model name here",
  "chip": "LN882H",
  "board": "TODO",
  "flags": "1152",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "0": "LED_n;0",
    "3": "Btn;0",
    "7": "BL0937CF1;0",
    "10": "WifiLED_n;0",
    "11": "Rel;0",
    "12": "BL0937CF;0",
    "19": "BL0937SEL;0"
  },
  "command": "",
  "image": "https://obrazki.elektroda.pl/YOUR_IMAGE.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic_YOUR_TOPIC.html"
}`

**To Reproduce**
Steps to reproduce the behavior:
1. Power up device, on homeassistant notice the correct mqtt data
2. wait ~ 2 hours
3. look again at homeassistant, notice all data is unkown, the uptime value stopped at ~2h
4. go to device web page, no response
5. look at router webpage , device offline
6. go to physical device, press button, no response
7. perform power cycle of device, now its running again.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
Add any other context about the problem here.

- the data was originally published every ~3seconds, which i found was alot. so i set command VCPPublishIntervals 60 60 to get updates every minute
- the device was flashed via uart following [this guide](https://www.elektroda.com/rtvforum/topic4028087.html#20905518) the only difference is i had to change the programming baud rate to 115200, a FW dump was also made prior to flashing
- i will attempt to leave the web application page open with the logs tab open to see if theres any usefull logs when the device becomes unresponsive

divadiow, 2024-03-25:
see this thread
https://www.elektroda.com/rtvforum/topic4044482.html

geniusled, 2024-03-25:
> see this thread https://www.elektroda.com/rtvforum/topic4044482.html

I just did a quick read through and i cant say for sure the issues are the same.
In that post is seems to focus on the webserver being slow after 30seconds and then being unresponsive. That is not my case.

The post does say they have a PR in progress so i will keep an eye out for a new release to see if there is any difference.

geniusled, 2024-03-27:
Hi just a quick feedback,
So the plug worked for about 20h then it froze the same way again.

I rebooted it and its been up for a few hours, ill report if it crashes again.

geniusled, 2024-04-01:
the device has been up for a few days now with 1.17.522, so it seems the original issue has been resolved. it seems that the uptime sensor was a bit slow updating since then but thats not an issue for me.
i just updated to 1.17.529 , ill see if this changes.

ill close this issue for now. Thanks you for the efforts of this project.

geniusled, 2024-04-06:
Sorry to bring this up again.
The device lasted for 2 days until it crashed again on 1.17.529

The device is inconveniently monitoring all my network stuff (nas,wifi,router,etc) so its quite inconvenient to power cycle it/reset.

would there be an option for a long press reset from the button?

geniusled, 2024-10-11:
Devices have been relatively stable since, closing

closed 2024-10-11 by geniusled