openshwprojects/OpenBK7231T_App #1153: IRsend command not sending anything

davidchezmoa, 2024-03-26:
**Describe the bug**
Cannot Send IR command, I got a cmnd not valid message but I don't know what could be the correct command.

**Firmware:**
- Version [1.17.308]
- Device IR sender 
- Chip/model: BK7231N
- Device config IR send pin 7

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '(http://IP/cmd_tool)'
2. Enter the command 'IRSend {"Protocol":"NEC","Bits":32,"Data":0x8166817E}'
3. Click 'Submit'
4. See error

**Error:IR:IRSend cmnd not valid [{"Protocol":"NEC","] not like [NEC-0-1A] or [NEC 0 1A 1].
Bad argument**

**Additional context**
If I do the exact same thing on a real Tasmota device it just work.

I didn't find any documentation on how the IRSend command work on OBK so I asume it work the same way as on Tamota ESP based device.

Thank you for your help and I will be happy to provide more information if needed.

openshwprojects, 2024-03-26:
Hey, I think you should try the syntax from the error message. We don't h ave JSON parse there yet, altough I may add it for you...

davidchezmoa, 2024-03-26:
I did try to mess around with the format in the error message, but I did not quite understand the way I should convert my initial data from my JSON.

If you have a hint it would be nice, otherwise I guess I will wait the implementation of JSON parsing 👍

openshwprojects, 2024-03-27:
Have you tried to check the docs?
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/commands.md