openshwprojects/OpenBK7231T_App #1156: Separate measuremente for negative power values

brunohoff, 2024-03-30:
Hello, 

There is the Flag 25 that enables negative values on power measurement and it works perfectly reporting current and power with positive and negative values. In my case I use a circuit break with power meter on the house and have solar panels generating energy. When the system is generating more power than the house usage, the power measurement is negative and reduces the total energy over time as expected and the opposite happens when is positive. 

The Energy Total I believe there is room for improvement because it is a more precise value than multiple snapshots from power usage over time to calculate KWh usage with integral or other methods. The suggestion is to create a separate measurement for the total amount of negative values for the day.

For example, the house consumed 10 KWh on the day and generated 9 KWh with solar panels. The OpenBK will correctly inform the energy for the day at 1KWh. For Home Assistant there is the option to inform the amount of energy generated with solar, the energy consumed by the house from the grid and the energy sent to the grid.

In this case it could be something like:
Energy Today: 1KWh
Negative Energy Today: 9KWh

Therefore it will be possible to calculate the energy of all the system.

Thank you in advance.

2Fblob, 2024-03-30:
I've been working on this for a few weeks now. See the thread:
https://www.elektroda.com/rtvforum/topic4045136.html

brunohoff, 2024-04-15:
Hello @2Fblob, that`s exactly what I am looking for. Do you need any help with the development? I can help you with the merging and pull request.

2Fblob, 2024-04-15:
Hi @brunohoff ,
You may want to try my fork. https://github.com/2Fblob/OpenBK7231T_App. The code is in a functional state. 

I'm having some problem with the WiFi routines - It will either recover itself and send the measurements on the next reporting period or it will restart. When it restarts it doesn't always connect to the router, but restaring the wifi on the router solves it. I have asked in the foruns and got no feedback if OB has any issues with the WiFi. So it would be good to have someone else testing :-)

I'm setting up another device on a different location to also run some tests.
I can make a build for you with some aditional parts of the code removed (so it only reports power measurements). 
Please ping me on the forums.

brunohoff, 2024-04-16:
Hi @2Fblob,

My only device is connected to my house main circuit breaker. I need a more stable version to flash into it. I purchased a second unit to test in a separate environment.

Did you tried without those hardcoded http request?
```
`CMD_ExecuteCommand("SendGet http://192.168.8.164/cm?cmnd=Power%20on", 0);
CMD_ExecuteCommand("setChannel 1 1", 0); 
```

You may use rules to power on or off other device. I usually use Nodered for those kinds of complex rules.

brunohoff, 2024-04-17:
Hello @2Fblob 

I created a separate code based on your on the repository: https://github.com/brunohoff/OpenBK7231T_App/
Please, feel free to test it and improve the code. 

I also created the PR https://github.com/openshwprojects/OpenBK7231T_App/pull/1184 so more people can help with the code.

2Fblob, 2024-04-17:
Hi,

Thanks. What features do you need? Netmetering? Or just two counters for Export / Import? We can remove anything not used.
The dual count functionality has been merged into the main branch a few weeks ago. That has a single counter that goes up and down based on the sum of consumption and export, when the negative flag is enabled.
We need another flag on Open Becken to accomodate for meters that report export as import and vice versa. This is not an issue on consumption only devices (as the value is absolute), but it is when we are measuring bi-directional.

Meanwhile, home assistant get's a bit confused when counters go down (including not reporting, if certain thresholds are not met) I have added ram variables, which load the main counters in periodic intervals. This should also help a bit with flash usage. On the standard code, flash is updated every few watts which results in a significant wear if we are measuing whole house consumption values.

I sugest giving it another 2 weeks, while I do some bug checking and adjustments. Currently have two real systems running and I'm doing some changes based on the results.

brunohoff, 2024-04-17:
Hello @2Fblob,

I want to use the circuit break as the main measurement device for Home Assistant. Right now I make it with multiple sources that are not very accurate.

I'm running the version that I shared before on my device and its not working as I expected, however, it is running without any instabilities. 

I believe the major issue is understanding the amount of energy generated. From what I see on your solution you tried to understand the flow of the energy without success and used the power to invert the energy accordingly.

I explain better the issue on the PR asking for guides: https://github.com/openshwprojects/OpenBK7231T_App/pull/1184#issuecomment-2062177436