openshwprojects/OpenBK7231T_App #1157: Powersave might not be working on BL602

giedriuslt, 2024-03-30:
**Describe the bug**
Powersave might not be working on BL602

**Firmware:**
- Version [e.g. 1.17]
- Device: Any
- Chip/model: BL602
- Device config:

Power save command on BL602 end here in SDK current SDK for OpenBK

> int wifi_mgmr_api_fw_powersaving(int mode)
> {
>     return wifi_mgmr_api_common_msg(WIFI_MGMR_EVENT_FW_POWERSAVING, (void*)0x1, (void*)0x2);
> }

So it does not depend on mode parameter and does something, but can't be hanged. Official sdk has fixed this.

divadiow, 2024-03-31:
On the topic of updated SDKs and big fixes in general, do the SDKs in this whole project get updated? Could there be other bugs and optimisations that have been made that would benefit OBK? Like the commits we see in the LN882H gitee.

divadiow, 2024-04-11:
is this an easy fix?

divadiow, 2024-04-11:
Probably a silly question but is moving to the latest SDK entirely a big task?

giedriuslt, 2024-04-11:
latest sdk has quite some changes in the structure, so app needs to be modified. I tried, but got stuck. I might return to that some time later

divadiow, 2024-04-26:
I wonder if the Ai-Thinker SDK is better. https://github.com/Ai-Thinker-Open/Ai-Thinker-WB2