openshwprojects/OpenBK7231T_App #1164: BL0942

Alexej550271, 2024-04-03:
Добрый день уважаемые разработчики! Скажите пожалуйста, вас не затруднит добавить поддержку драйвера BL0942? Очень нужно... Думал, что она уже есть, но в модулях нет такого выбора, для подключения.

divadiow, 2024-04-05:
OBK has support for BL0942

openshwprojects, 2024-04-05:
BL0942 is supported along with MQTT hass pairing:
https://www.youtube.com/watch?v=pkcspey25V4

Alexej550271, 2024-04-06:
Большое спасибо, что уделили мне ваше драгоценное время и ответили на мой
глупый вопрос. Как я теперь уже понимаю в системе есть драйвера и есть
команды для их включения, но не могу найти где можно почитать об этом
подробнее, ткните пожалуйста носом, подскажите пожалуйста нужную ссылку,
где эта вся информация описана. Дело в том, что мне попался "новый девайс"
который ещё никто не прошивал или я просто этого не смог найти..
Автоматический (умный) выключатель на дин рейку (1дин), Немного помучившись
(до сих пор разбираюсь) Мне удалось его заставить работать как
выключатель/включатель из веб интерфейса, также вроде бы удалось подключить
драйвер bl0942, но есть две проблемы которые я пока не могу решить...
Первая, не знаю что, где и как прописать, чтобы он также включался с кнопки
на его корпусе, дело в том, что реле переключается через мост вперёд и мост
назад, это реле мне очень понравилось этим, оно просто передвигает
выключатель в нужное положение "включено/выключено" и находится без
напряжения на катушке. Вторая проблема данные с "датчика" bl0942 не
обновляются в веб интерфейсе о не отправляются по mqtt, в веб интерфейсе
постоянно только нули.. Помогите пожалуйста настроить и у нас будет ещё
один готовый "девайс", я вам скину фото, распиновку и прочее, что
потребуется для того, чтобы его добавить. А также дайте пожалуйста ссылки
где можно подробно почитать как настраивать устройства, как активировать
драйвера и прочее, а то очень много всего читал, смотрел, перерыл всё, что
попадается, но нужной информации всё ещё не удалось найти, я очень сильно
уже запутался и у меня скоро взорвётся голова))) Очень надеюсь на вашу
помощь. Заранее огромное вам спасибо! С огромным уважением за ваш
проделанный труд!!!

Alexej550271, 2024-04-06:
И ещё один момент забыл указать, я использую эти устройства не с "НА", а с
другим проектом "iotmanager", через "костыли", потому мне нужен просто
mqtt. Пару розеток (bl602 и bk7231) с драйверами bl0937, мне удалось
привязать и всё нормально работает, а вот bl0942/bk7231n, настроить не
получается в вебе вроде бы он появился, но похоже, что не работает...
Совершенно все данные с него в веб интерфейсе, в том числе напряжение он
показывает только 0.00, а в mqtt, не отправляются даже эти нули.. И как
ранее писал не могу понять как настроить чтобы с кнопки на корпусе
переключался мост, что и где нужно прописать, буду вам очень признателен,
если поможете решить эти проблемы. Или хотя бы напишите ссылки, где можно
найти инструкции по настройке устройств и драйверов, а также команды их
включения.
   С уважением Алексей.