openshwprojects/OpenBK7231T_App #1166: Version number is not updated on TuyaMCU devices

wimoos, 2024-04-04:
**Describe the bug**
After OTA upgrade, the new version in not reflected - was the upgrade successful ?

**Firmware:**
- Version: 1.7.521 and up
- Device: behaviour seen with MS-105 dimmer and with Wifi power measurement device
- Chip/model: BK7231N
- Device config: TuyaMCU

**To Reproduce**
Perform OTA and reboot through Web application OTA
After reboot, check version number on main screen to find it has not changed

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

openshwprojects, 2024-04-04:
I can stop all drivers before executing OTA, do you think that would help?

Did it also happen in older versions?

wimoos, 2024-04-04:
It occurred to me while upgrading to 1.7.521. Can't remember anymore from what version.

I only use TuyaMCU as driver in some of my projects , so I can't say if this problem would occur with other drivers.
As a whole it is indeed maybe better to stop (or suspend) all drivers while doing the OTA.

openshwprojects, 2024-04-04:
It's not completed yet, I haven't added a fix yet.

wimoos, 2024-04-05:
My bad, sorry. I hit the wrong button.
And the version number was 1.**17**.521, of course :-) .

Keep up the good work!