openshwprojects/OpenBK7231T_App #1174: Send constant with PublishFloat/PublishInt/Publish on BL602

Flobul, 2024-04-07:
**Describe the bug**
Unable to send constant values to MQTT, values sent are always 0.
Tried with PublishFloat, PublishInt and Publish.

But works with plain numbers. (see screenshot)
Works fine on BK7231N device.

**Firmware:**
- Version 1.17.538
- Device : https://www.elektroda.com/rtvforum/topic4020634.html
- Chip/model: BL602
- Device config:
```
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Full Device Name Here",
  "model": "enter short model name here",
  "chip": "BL602",
  "board": "TODO",
  "flags": "69731332",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "2": "Rel;1",
    "3": "BL0937CF1;0",
    "14": "BL0937CF;0",
    "17": "Btn;1",
    "20": "LED_n;1",
    "21": "BL0937SEL;0"
  },
  "command": "backlog startDriver BL0937; startDriver NTP; ntp_setServer 82.64.230.205; ntp_timeZoneOfs 1:00; addRepeatingEvent 5 -1 publishFloat Temperature $intTemp",
  "image": "https://obrazki.elektroda.pl/9568515100_1707749274.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic4020634.html"
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Send command "publishFloat TEST $intTemp" or any other constant
2. Check on MQTT and see "0"

**Screenshots**
[image]
[image]
[image]
[image]

Flobul, 2024-04-21:
Ok, done, but not as relevant as I thought.
[image]
[image]
[image]

Same for other $uptime, day, hour...

giedriuslt, 2024-05-12:
That is actually a buf, hopefully will be fixed in https://github.com/openshwprojects/OpenBK7231T_App/pull/1229

Flobul, 2024-05-27:
Fixed for me :) thanks