openshwprojects/OpenBK7231T_App #1177: BL602 devices crash and reboot when fingered with port scanners

divadiow, 2024-04-11:
**Describe the bug**
MAIN ISSUE: I've tried Windows NMAP, Advanced Port Scanner and Angry IP Scanner, targeting port 80 specifically and separately, all ports. This has been done on two different machines - cabled main PC and virgin laptop on wifi. Each scan will cause BL602 watchdog to eventually reboot the device. Fing Android app on mobile does not have the same effect.

SECONDARY ISSUE: ICMP ping responses from BL devices are either non existent or intermittent. NOT an issue with wifi strength - different APs tested. http gui is still available on port despite no ping response, device remains online through ping attempts.

**Firmware:**
- Version Apr 10 2024 13:59:03 version 1.17.542
- any BL602 - dev board, SM-028 Qiachip relay and LF686 RGBW LED controller tested
- Chip/model: BL602
- Device config: fresh BL flash with and without any drivers - no difference

**To Reproduce**
Steps to reproduce the behavior:
1. Boot up BL602 device and connect to WLAN
2. scan for ports or just port 80
3. 
4. BL602 will crash and reboot. Reboot reason: BL_RST_SOFTWARE_WATCHDOG

maybe this isn't such a big issue - just don't port scan! my LN and BK devices disconnected wifi on scan (but then reconnected themselves) but did not crash/reboot, so BL is more sensitive.

dev board uart output at time of port scan:

```
[21:36:04.930] - , free 106728, MQTT 0(14), bWifi 1, secondsWithNoPing 165, socks 2/2
[21:36:04.931] - 1
[21:36:05.932] - Info:MAIN:Time 235, idle 0/s, fr
[21:36:05.933] - ee 106728, MQTT 0(14), bWifi 1, secondsWithNoPing 166, socks 2/2
[21:36:05.934] - 1
[21:36:06.932] - Info:MQTT:mqtt_host empty, not s
[21:36:06.936] - tarting mqtt
Info:MAIN:Time 236, idle 0/s, free 106728, MQTT 0(
[21:36:06.937] - 15), bWifi 1, secondsWithNoPing
[21:36:06.943] -  167, socks 2/21
[21:36:07.939] - Info:MAIN:Time 237, idle 0/s, fr
[21:36:07.940] - ee 106728, MQTT 0(15), bWifi 1, secondsWithNoPing 168, socks 2/2
[21:36:07.940] - 1
[21:36:08.931] - Error:HTTP:TCP Client is disconn
[21:36:08.935] - ected, fd: 2
[21:36:08.942] - Info:MAIN:Time 238, idle 0/s, fr
[21:36:08.944] - ee 106440, MQTT 0(15), bWifi 1, secondsWithNoPing 169, socks 3/2
[21:36:08.945] - 1
[21:36:09.181] - Current t
[21:36:09.187] - ask sp data:
RA:23031e78, mstatus:80007880
A0:00000000 A1:00000d0d A2:1f1f1f1f A3:0000000d A4:fefefeff A5:fefefeff A6
[21:36:09.189] - :fefefeff A7:80808080
T0:2302fab0 T1:20202020 T2:a5a5a5a5 T3:a5a5a5a5 T4:a5a5a5a5 T5:a5a5a5a5 T6:a5a5a5a5
S0:42029ee4 S1:230baa68 S2:4202a880 S3:00000004 S4:230adcd4 S5:00000004 S6:a5a5a5a5 S7:a5a5a5a5
S8:a5a5a5a5 S9:a5a5a5a5 S10:a5a5a5a5 S11:a5a5a5a5
[21:36:09.190] - Exception Entry--->>>
mcause 38000005, mepc 2308b910, mtval 00000000
Exception code: 5
msg: Load access fault
[21:36:11.874] - Starting bl602 now....
Booting
```

issue starts at **[21:36:08.931] - Error:HTTP:TCP Client is disconn**

divadiow, 2024-04-16:
hmm. no specific commands, theyre just gui scanners. nmap was intense profile. not sure of the mechanics beneath these scans and how they differ or why Fing on Android didnt have the same effect
[image]

giedriuslt, 2024-05-11:
Can you check if It crashes with latest release (from github actions)

divadiow, 2024-05-11:
nice one. no crashes with nmap/angryip/advanced port scanners. just this in log now

[image]