openshwprojects/OpenBK7231T_App #1182: Problem after restart host computer

rafalolb, 2024-04-16:
Hello, 

I have 5 switch with installed OpenBK firmware. I using Home assistant with MQTT installed in Dockers on my RPI. All switches works perfect, but I found 1 problem.

If I restart my host (my RPI), all my switches are unavaible. I must go to Config->Home Assistant Configuration and click Home assistant discovery. If I do this everything works great.

Why I must click discovery if I restart my host computer? Can I do something (for example set some flag)? 

Please help.

openshwprojects, 2024-04-16:
Hello, do you have set "publish self state every N seconds" or how it was called, flag? 

If you restart your host (RPI), and then toggle value of a switch, does it also gets discovered?

Or, to be clear, are they removed from config (so you have to rediscover) or maybe it's just that they are shown as unavailable?

rafalolb, 2024-04-17:
> Hello, do you have set "publish self state every N seconds" or how it was called, flag?
> 
Yes, I have check checkbox Flag 2 - [MQTT] Broadcast self state every N (def: 60) seconds (delay configurable by 'mqtt_broadcastInterval' and 'mqtt_broadcastItemsPerSec' commands)

> If you restart your host (RPI), and then toggle value of a switch, does it also gets discovered?
> 
If I toogle value of state on switch UI, the state of switch will be unavailable on Home Assistant. 

> Or, to be clear, are they removed from config (so you have to rediscover) or maybe it's just that they are shown as unavailable?
I don't understand. I don't have configuration in configuration.yml, but the switches exists in Settings -> Integration -> MQTT, but all switches are unavailable.

rafalolb, 2024-04-17:
I have installed MQTT on RPI host. I think that, if MQTT server is available for some time, after restart the switch not reconnect to mqtt, but on UI I see connected to MQTT.

rafalolb, 2024-04-21:
I have additional info. If I turn off my router and next turn on for some minutes, switches works. Problem only exists when host (RPI) is restarted.

pikpok, 2024-05-02:
I had a similar issue, and setting flag 21 did the trick for me:
```
Flag 21 - [MQTT] Retain power channels (Relay channels, etc)
```

rafalolb, 2024-05-02:
Doesn't works for me 😒

jonny190, 2024-05-16:
I'm facing the same issue, currently looking at a Home assistant automation that restarts the devices when they show as "unavailable" if the system uptime is less then 5 mins. Its more of a work around as i'm not sure what else to do at present

slo617, 2024-05-19:
I have had the same problem, `Flag 27 - [HASS] Invoke HomeAssistant discovery on change to ip address, configuration` helped me.

Once published discovery information is stored on the MQTT-broker. If the MQTT-broker (and Home-Assistant) restarts this information is lost.
So we need to to re-send discovery information every time we reconnect to the MQTT-broker.

Actually, `change to ip address, configuration` and _reconnect to the MQTT-broker_ is not exactly the same, so either flag 27 might not help in all cases or documentation should be improved.

slo617, 2024-05-19:
I guess flag 27 worked for me because I got home assistant running on my router. Restarting MQTT broker and home assistant usually means restarting the whole router for me.
So this is not a generic solution.

As a workaround you could start home assistant discovery from time to time by a script, e.g. in `autoexec.bat`. You could try something like this:
```
addRepeatingEvent 10 -1 scheduleHADiscovery 5
```
*Be warned, I have not tested it.* It should schedule a discovery every 10 seconds with a delay of 5 seconds.
If it works, it will also add a very little bit of traffic to your WLAN. And there will be a delay of up to 10 seconds.
See also: https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/commands.md

Better solution would be an event that is fired by the MQTT driver, or a modification of flag 27, so it sends discovery information on re-connection to the MQTT broker.
So this issue can still be considered as a bug, or at least as a feature request.

bilogic, 2024-10-16:
https://github.com/openshwprojects/OpenBK7231T_App/issues

yes, it should be classified as a bug

bilogic, 2024-11-10:
I tested out ESPHome and find that it is much more reliable and probably more suitable for github users who should be a bit more code savvy.

FeelTheLemon, 2025-04-28:
Hi @openshwprojects
Any plans on fixing this problem?