openshwprojects/OpenBK7231T_App #1183: faulty get payload to 1 when should be 0

Magnus-rosenborg, 2024-04-16:
**Describe the bug**
I recieve mqtt messages with payload 1 when it is 0. I have set Flag 7 - [MQTT] [Debug] Always set Retain flag to all published values
but the messages are not retained. When i send an emty mqtt like mosquitto_put -t 37/1/get -m "" -u user -P password it respond correct with payload 0. But after som minute it sends the wrong mqtt again.

**Firmware:**
- Version 1.17.546
- Device? 3 button touch wall switch
- Chip/model: BK7231N
- Device config?

autoexec.bat
startDriver ntp
ntp_setServer 193.11.166.8
ntp_setLatlong 56.017833, 14.268284
AddChannel 10 1
AddChangeHandler Channel4 > 0 publish 37/KNAPP1 TOGGLE -1
AddChangeHandler Channel5 > 0 publish cmnd/KNAPP113 TOGGLE -1
AddChangeHandler Channel6 > 0 publish 37/KNAPP3 TOGGLE -1

FLAGS: "flags": "1075840224",
 "pins": {
    "6": "Rel;1",
    "7": "dInput_n;6",
    "8": "Rel;2",
    "9": "Rel;3",
    "11": "Rel_n;3",
    "14": "Rel_n;1",
    "20": "dInput_n;5",
    "22": "Rel_n;10",
    "24": "dInput_n;4",
    "26": "Rel_n;7"
  },

**To Reproduce**
note, it repeats every 3 minute.

**Screenshots**
[image]
[image]
[image]

**Additional context**
Add any other context about the problem here.

Magnus-rosenborg, 2024-04-16:
The chip inside is a CB3S if it is helping?

openshwprojects, 2024-04-16:
Hey, why do you have Rel_n roles? Aren't you... just inverting the value yourself?

Magnus-rosenborg, 2024-04-17:
Hi, thats how it's constructed. Everything else is working correctt, but this faulty mqtt messages that is generated inside is wrong 0 means off etc. the relay is of when the device is sending 1. When i query the device it respond correct with 0.

openshwprojects, 2024-04-18:
You have both Rel and Rel_n on channel 1, does the problem persists if you remove temporarily one of them?

Magnus-rosenborg, 2024-04-26:
Hello! Sorry for not answering (got sick) and the hardware is broken now. I believe there were something wrong with that device only.

closed 2024-04-26 by Magnus-rosenborg