openshwprojects/OpenBK7231T_App #1188: Enhacement: Internal temperature Home Assistant discovery

f-vicente, 2024-04-22:
**Describe the bug**
An enhacement related with https://github.com/openshwprojects/OpenBK7231T_App/issues/1125

Will be great a flag that enables the internal temperature to be published, and generates the appropriate HASS discovery.

At this moment you need to create the timmer to publish the variable value (like shown on #1125), then create the appropriate sensor manually in home assistant. Will be great if this could be automated.

**Firmware:**
- Version 1.17.553
- Chip/model: BK7231N

openshwprojects, 2024-04-23:
Sure, it's planned and I can add it for you, but first, are you aware that this internal temperature is not a perfect measurement and should not be really relied upon? As far as I know it' even has serious issue on one of the platforms. The fix is still pending.

f-vicente, 2024-04-24:
Great!

Yes I know that the sensor is very inaccurate, but I like to have statistics about it, and use the values to trigger alarms if they get very high values that could reflect any problem. I only have bk7231n devices and those seem to work reasonably fine.

pablacho, 2024-05-29:
I have the latest version 1.17.596, rediscovered all my lights so the new sensors could be added. I can now see on each light the Build/RSSI/Temperature/Uptime sensors in the Diagnostics tab, but they all stay in 'Unknown' state. I have tried restarting the lights and even sending the "addRepeatingEvent 5 -1 publishFloat SelfTemperature $intTemp" command but no effect. 

Is there anything else I need to do for the values to update on my HA?

f-vicente, 2024-05-29:
Thanks for adding it.

f-vicente, 2024-05-29:
> Hi! I am on the latest version of OpenBK7231T. I have re-discovered all my devices to add the new sensors, but they only update on restart! should I add repeating events for each sensor? what are their variable names? I am asuming $intTemp should go to topic 'temp', (so running "addRepeatingEvent 300 -1 publishFloat temp $intTemp" should do the trick), but what are the others?

I haven’t done anything in special, just update and the new sensors appeared.

closed 2024-05-29 by f-vicente

pablacho, 2024-05-29:
> I haven’t done anything in special, just update and the new sensors appeared.

> I haven’t done anything in special, just update and the new sensors appeared.

> I haven’t done anything in special, just update and the new sensors appeared.

do they update every now and then? Mine stay the same (14mins and counting), I am checking the MQTT log but no messages arrive. I have also tried running the command I suggested but the info goes to <dev_topic>/temp/get instead of <dev_topic>/temp so HA ignores it...

f-vicente, 2024-05-29:
> do they update every now and then? Mine stay the same (14mins and counting), I am checking the MQTT log but no messages arrive. I have also tried running the command I suggested but the info goes to [...]

Yes, the update as expected.

I have experienced the issue on one of them, it didn’t show anything, but it was fixed when I disconnected from power and connected again. I have a few devices that always give me those kind of errors after the update.

Phobos-7, 2024-11-05:
So this is super cool:
[image]
But, it's getting updated only once - on startup :(
Is there a way to make it regular, every 15 seconds for example?

Phobos-7, 2024-11-05:
Oh, my bad. It can be set here:

Config - Configure General/Flags
Flag 2 - [MQTT] Broadcast self state every N (def: 60) seconds (delay configurable by 'mqtt_broadcastInterval' and 'mqtt_broadcastItemsPerSec' commands)