openshwprojects/OpenBK7231T_App #1190: Unable to control LED dimmer, channel value rises absurdly high by itself

tobiaspc, 2024-04-30:
**Describe the bug**
LED dimmer can not be controlled after some time, since the dimmer value rises to absurd values. "Smooth transitions for LED" (Flag 18) is enabled.

**Firmware:**
- Version 1.17.555
- Device: Magic Home single channel Wifi LED controller
- Chip: BL602
- Device config:
Default, using MQTT, Flags 10, 12, 18 active

**To Reproduce**
Steps to reproduce the behavior:
1. Power on, use / toggle the light normally
2. After extended power on time (days), LED channel value rises to absurd values (i.e., over 100.00), see first screenshot.
3. Cut the power, turn on again
4. Value rises very fast, exceeding the upper limit of 100 (second screenshot)
5. Software thinks light is "off" (red button color), despite it being already (physically) on
6. Press Toggle Light button
7. Press Toggle Light button again, LED dimmer value decreases, reaching 0 after some time.
8. From there on, normal operation, i.e., LED values are clamped at 0 and 100.

This bug is not deterministic, sometimes it appears after a few hours, sometimes after days. Manually setting the brightness to 0 or 100 does not work, neither via "normal" web-UI, nor via "Web Application". Only way to workaround is cutting the power and quickly toggling the light twice, as described above.

**Screenshots**
Bug during operation:
[image]

Bug after reboot:
[image]

**Additional context**

openshwprojects, 2024-04-30:
Hello, can you try to narrow down the issue, what happens if you try the same without smooth led transitions, etc?

tobiaspc, 2024-05-01:
Thanks for the idea. Just disabled Flag 18, will see how it behaves over the next few days and report back.

tobiaspc, 2024-05-05:
The issue did not occur since I turned off the smooth transition feature, so I guess this is the cause. Any idea on how I could debug this further?