openshwprojects/OpenBK7231T_App #1206: Online builds are failing with Error: Cannot find module 'conventional-changelog-conventionalcommits'

openshwprojects, 2024-05-04:
Ideas? It seems I need to look into it soon. Maybe they updated the package.

Wolfieeewolf, 2024-05-06:
Not sure if this helps

https://github.com/conventional-changelog/commitlint/issues/613

looks like it might be an issue. towards the bottom, it looks like there are some workarounds.

giedriuslt, 2024-05-06:
I see we use semantic-release-action@v3 but v4 is latest? maybe upgrade would help. Also other actions should be upgraded they produce warnings

giedriuslt, 2024-05-06:
According to docs here https://github.com/marketplace/actions/action-for-semantic-release#extra_plugins
extra plugings should also be in releaserc file.
That conventional-changelog-conventionalcommits is missing in there

DeDaMrAzR, 2024-05-12:
@openshwprojects 

I tried a solution on a fork of mine and it seems to work, semantic-release needs to be pushed to v21 along with several other plugins. I've sent you a file privately for testing, not eager to brake anything public :)

closed 2024-05-13 by openshwprojects