openshwprojects/OpenBK7231T_App #1217: Combined BMP280/AHT20 sensor module not working with BMP280 PR

divadiow, 2024-05-07:
I've a CB3S based dev board. 2 SimpleRobot BMP280/AHT20 combined i2c modules from Ali Express. 1 has a non-functioning AHT20.

The modules are being connected to P20 - SCL and P21 - SDA. Both are detected correctly by "softscan i2c" as

**0x38 (dec 56)** and **0x38 (dec 56), 0x77 (dec 119)** (fully functioning module).

The i2c address behaviour and sensor function is tested as working in Tasmota on an ESP device. Same addresses detected as above for each module.

In OpenBeken running PR OpenBK7231T_App_1111_merge_d5110564bd99 from [here](https://github.com/openshwprojects/OpenBK7231T_App/actions/runs/8144046764?pr=1111)
the "startdriver bmp280 21 20 1 2" and "startdriver bmp280 21 20 1 2 119" commands always return 

```
Info:SENSOR:BMx280 wrong ID!
Info:SENSOR:BMP280 failed!
```
It took me a while to realise the CHIP ID and I2C address were different things but the BMP280 is correctly detecting as 0x58 in Arduino detection sketch on an ESP8266 with this result

```
Could not find a valid BME280 sensor, check wiring, address, sensor ID!
**SensorID was: 0x58**
        ID of 0xFF probably means a bad address, a BMP 180 or BMP 085
        ID of 0x56-0x58 represents a BMP 280,
        ID of 0x60 represents a BME 280.
        ID of 0x61 represents a BME 680.
```
taken from https://forum.arduino.cc/t/solved-i2c-bme280-bmp280-works-with-bmp-begin-0x76-0x58/689445

The bmp280.h driver is set for 0x58 for the BMP280 https://github.com/openshwprojects/OpenBK7231T_App/blob/testBMP/src/driver/BMP280.h

[image]

The driver was not telling me the actual flash ID detected though so I tried this PR which should add detected flash ID to the output https://github.com/openshwprojects/OpenBK7231T_App/pull/1213/commits/cff72e62c8480dbc19f103a660862d84b79e84cc

it compiled and it gave an output. If the driver change worked it was reading the flash ID at 0xFF, no wonder it was saying wrong ID!

```
Info:SENSOR:Read chip ID: FF
Info:SENSOR:BMx280 wrong ID!
Info:SENSOR:BMP280 failed!
```
I also went back and tried the original 1.17.361 release from 17th December from the original thread [here](https://www.elektroda.com/rtvforum/topic4021224.html) but no change to function. Apart from the user having a BM**E**280 and using a CB2S board (looks like D0/P14 D1/P8 are used) I'm not sure what the functional difference is. I have also tried D0/P14 and D1/P7 and then a different pair on RX/P10 and TX/P11 with no luck. on all pin pairs tried the i2c addresses were detected as in my first few lines.

so, why isn't the chip ID being detected in my case?

does anyone have a standalone BME/BMP280 to check again? does the flash ID report PR give anything other than FF?

on a separate note I have tried a single BME680 module and the CHIP ID is also not read. I also made a PR substituting the BME280 for 0x61 of the BME680 but this produces the same wrong ID and no working sensor.

divadiow, 2024-05-08:
Mentioning @NonPIayerCharacter who has the same module if able to help

NonPIayerCharacter, 2024-05-08:
I used startdriver bmp280 8 7 3 4 238, where 238 == 0x77 << 1

divadiow, 2024-05-09:
omg. thank you. I've spent ages playing this!

I had to ask ChatGPT was the relationship of value "238" was in the context of 0x77, 119 and why. I had no idea what a left bit-shift was!

[image]

and this shows my little addition to the bmp280 driver also works :) very happy now. thanks

divadiow, 2024-05-09:
[image]

kind of feel like making a new thread on Elektroda dedicated to it like the AHT2X driver one

closed 2024-05-09 by divadiow