openshwprojects/OpenBK7231T_App #1219: BL602 - waitFor NTPState 1 doesn't seem to work and waits forever.

delay0, 2024-05-08:
"waitFor NTPState 1" doesn't seem to work and waits forever.

running Build: TMGN on beken #1218 ( https://github.com/openshwprojects/OpenBK7231T_App/pull/1218 )

  "vendor": "Tuya",
  "name": "Surplife APP RGB+WW+CW bulb WiFi+BLE E27",
  "chip": "BL602",
  "board": "BL-M5 V1.2",

delay0, 2024-05-08:
in cmd_eventHandlers.c

```
302 | #if defined(PLATFORM_BEKEN) \|\| defined(WINDOWS)
--
303 | CMD_Script_ProcessWaitersForEvent(eventCode, newValue);
304 | #endif
 
386 | #if defined(PLATFORM_BEKEN) \|\| defined(WINDOWS)
387 | CMD_Script_ProcessWaitersForEvent(eventCode, argument);
388 | #endif
```

I released and tested this for my BL602. Works with NTPState.

giedriuslt, 2024-05-08:
Fixed here #1191 not sure when will be merged or builds fixed