openshwprojects/OpenBK7231T_App #1220: bl602 - Problems with saving, editing and executing scripts

delay0, 2024-05-08:
I'm starting to feel sorry for throwing so much into this, but this is something I've been dealing with for days. My English is not good but I'll try to explain.

In the file editor, something isn't working properly when saving, loading/executing.

Processes:
-Start controller
-Open file
-click "save and run"

sometimes the file is started, sometimes not. Clicking multiple times shows in the log that it is not working properly. after a few clicks the file will be executed. I can't tell whether it has something to do with time or the number of clicks, or whether it happens completely randomly and always differently.

-click "reset scripts"
sometimes helps.

Now (or directly after opening the file) I change the file.

-click "save and run"

Sometimes the file is executed, sometimes not. see above.
When executed, the old file version is used, not the newly saved one.

-click "reset scripts"

sometimes helps, sometimes not. see above.

last chance now...
-click the other button "save, reset (...) and run..."

Result: the controller restarts completely, the script is not executed.
I tried to temporarily fix the problem with a resetSVM at the end of the script. but even that only seems to help sometimes.

Additionally:
I once had files disappear and old ones come back. (files created and deleted within one session.).

And finally... edited file could not be saved. I had to delete them and recreate them.

The last two problems have only occurred once so far

  V1.17.571 released and last builds eg. TMGN on beken https://github.com/openshwprojects/OpenBK7231T_App/pull/1218 
  "vendor": "Tuya",
  "name": "Surplife APP RGB+WW+CW bulb WiFi+BLE E27",
  "chip": "BL602",
  "board": "BL-M5 V1.2"

giedriuslt, 2024-05-11:
Cannot reproduce the issue with sometimes script is started, but I see that after save old version is run. Will try to fix

giedriuslt, 2024-05-12:
Did some investigation. 
Save, Run.. always run original version, that's the same on all platforms
Reset scripts - does not work due to bug, fix waiting here https://github.com/OpenBekenIOT/webapp/pull/111
Save, Reset and run crashes bl602 chip, fix is waiting here https://github.com/openshwprojects/OpenBK7231T_App/pull/1226