openshwprojects/OpenBK7231T_App #1222: Selecting "Configure Module" menu option causes device to lose connection.

soundfx42, 2024-05-08:
**Describe the bug**
When accessing device via builtin webpage it will become unavailable when selecting the "Configure Module" menu. The device will suddenly just stop responding with a "connection has timed out" error. Device will also no longer respond to ping requests with "Destination host unreachable" response. The only way to recover is to power cycle the device.

Utilizing the Pin Watchdog function allows the device to recover without the need to power cycle the device however the Configure Module menu is still not accessible as the connection to the device will immediately time out once again if that menu is selected.

**Firmware:**
- Version: 1.17.262
- Device: Arlec Grid Connect Smart Plug PC191HA
- Chip/model: BK7231T

**To Reproduce**
- Access device via browser (Firefox) using IP address of device.
- Select "Config"
- Select "Configure Module"

**Additional context**
I currently have three of these devices installed. They all display the same behavior. The devices can be configured by using the "Launch Web Application" option.

openshwprojects, 2024-05-11:
Interesting, I have managed to get the same problem on Firefox, while it works for me on chrome... futher investigation needed.

soundfx42, 2024-05-11:
I tried it in Chrome and the Configure page loaded the first time but now every time after it only partially loads before hanging.
[image]

I disabled all extensions and cleared everything in the cache and tried again but again, only got the partially loaded page.

openshwprojects, 2024-05-11:
Strange, it doesan't manifest so strongly for me:
[image]
Still, I will look into it, I may also think how to just shorten the list of available io roles because currently this list is  printed to html for each dropdown...

openshwprojects, 2024-05-11:
imworking on it https://github.com/openshwprojects/OpenBK7231T_App/commit/b3db6df997259176708414f80e031ad7e7190c83

openshwprojects, 2024-05-12:
fixed, i just need to fix main releases stystem
https://www.elektroda.com/rtvforum/topic4053623.html

soundfx42, 2024-05-13:
I would love to test this out the the problematic devices but I need some direction on how to go about it. 

The original jailbreak was done over wifi with tuya-cloudcutter. I have since updated the firmware on the device to 1.17.571 OTA using the Web Application. 

That is the extend of my knowledge unfortunately. If you could instruct me on how to install the testbuild with the fix I would happily have a shot at testing it for you.

DeDaMrAzR, 2024-05-13:
@soundfx42 

Can you try now on the latest public build? Thanks

soundfx42, 2024-05-13:
I've upgraded one of the devices to 1.17.574 and it works like a charm. The Configure Module page loads instantly without any sign of timeouts. I will upgrade the other two modules over the next few days. Nice work on the prompt fix!