openshwprojects/OpenBK7231T_App #1237: SM16703P Driver causing device reboot to safe mode - 1.17.566

Wolfieeewolf, 2024-05-14:
**Describe the bug**

The device hangs once you toggle the light. Appears to continually reboot but the lights stay on. Have seen up to 200 reboot attempts. Have to reboot into safe mode to restore.

Have tried the latest firmware 1.17.581 and the issue is still there.

**Firmware:**
- Version 1.17.566

Lots of changes to the pixel driver happened in 1.17.566

- Device - [Arlec Grid Connect Smart Desktop Bar Lights](https://www.elektroda.com/rtvforum/topic4013211.html)
- Chip/model: BK7231N
- Device config 

No pins set

Flag 4 set 

LittleFS

autoexec.bat 

```
startDriver SM16703P
SM16703P_Init 30 GRB
again:
SM16703P_SetPixel all $led_enableAll*$led_red*$led_dimmer/255 $led_enableAll*$led_green*$led_dimmer/255 $led_enableAll*$led_blue*$led_dimmer/255
SM16703P_Start
delay_s 1
goto again
```

have also tried 

```
startDriver DDP
startDriver SM16703P
SM16703P_Init 30 GRB
```

Both settings have worked on 1.17.565 with no issues.

**To Reproduce**
Steps to reproduce the behavior:
1. OTA to '1.17.566'
2. Click on 'Toggle 1'
4. The device hangs LEDs stay on. Have to reboot to safe mode to fix

openshwprojects, 2024-05-14:
Thank you for reporting. Can you check if it's working with older versions? Which version broke SM16703P?

openshwprojects, 2024-05-14:
I think you don't need that script anymore. It's now integrated into OBK without script. That's why you are getting crash...

Wolfieeewolf, 2024-05-14:
Do I need to start the driver?

I just tried

startDriver SM16703P
SM16703P_Init 30 GRB

and it still crashes

Wolfieeewolf, 2024-05-14:
I deleted the autoexec.bat and it appears to be working now. 

Very strange. I did try that before but nothing would light up. 

For now, it's working so I'll go ahead and close this issue

closed 2024-05-14 by Wolfieeewolf