openshwprojects/OpenBK7231T_App #1247: Compile for BK7231T using cygwin - missing crt...

xjikka, 2024-05-24:
I'm trying to compile OpenBK7231T_App using Cygwin and I'm getting compile error (crt*.o).
Any help would be appreciated.

log:
d:/_dataprg/cygwin/bk7231t/platforms/bk7231t/toolchain/windows/gcc-arm-none-eabi-4_9-2015q1/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: cannot find crti.o: No such file or directory
d:/_dataprg/cygwin/bk7231t/platforms/bk7231t/toolchain/windows/gcc-arm-none-eabi-4_9-2015q1/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: cannot find crtbegin.o: No such file or directory
d:/_dataprg/cygwin/bk7231t/platforms/bk7231t/toolchain/windows/gcc-arm-none-eabi-4_9-2015q1/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: cannot find crt0.o: No such file or directory
collect2.exe: error: ld returned 1 exit status
make[2]: *** [application.mk:778: application] Error 1
make[2]: Leaving directory '/cygdrive/d/_DataPrg/cygwin/bk7231t/platforms/bk7231t/bk7231t_os'
make[1]: *** [Makefile:5: mp] Error 2
make[1]: Leaving directory '/cygdrive/d/_DataPrg/cygwin/bk7231t/platforms/bk7231t/bk7231t_os'
make: *** [Makefile:6: all] Error 2

xjikka, 2024-05-25:
I temporarily solved it by copying files from docker compiler mode. 
Anyway, I want to solve it right way. 
So, is there anyone compiling OpenBK7231T_App using Cygwin?

openshwprojects, 2024-05-29:
I am using cygwin setup as shown here:
https://www.elektroda.com/rtvforum/topic3850712.html

xjikka, 2024-05-29:
> I am using cygwin setup as shown here: https://www.elektroda.com/rtvforum/topic3850712.html

Thank You very much, I've tried it, but unfortunately without success. It ends up with this error:

../toolchain/gcc-arm-none-eabi-4_9-2015q1/bin/arm-none-eabi-gcc: ../toolchain/gcc-arm-none-eabi-4_9-2015q1/bin/arm-none-eabi-gcc: **cannot execute binary file**

but the file is there, and has permissions, execute it gives error:

xjw@NB-XXXL /cygdrive/d/a/tuya-iotos-embeded-sdk-wifi-ble-bk7231t/platforms/bk7231t/toolchain/gcc-arm-none-eabi-4_9-2015q1/bin
$ ./arm-none-eabi-gcc
-bash: ./arm-none-eabi-gcc: cannot execute binary file: Exec format error

I've tried to redownload but the file is same.

The only difference I see that folder Windows is missing (in compare to Your screenshot here https://www.elektroda.com/rtvforum/topic3850712.html)
[image]

terminal log:
sh build_app.sh apps/template_demo template_demo 1.0.0
APP_PATH=apps/template_demo
APP_NAME=template_demo
APP_VERSION=1.0.0
USER_CMD=
APP_BIN_NAME=template_demo
APP_VERSION=1.0.0
TARGET_PLATFORM=bk7231t
USER_CMD=
Start Compile
Removing TUYA APP Common and Components Objs
not is ci build
make: Entering directory '/cygdrive/d/a/tuya-iotos-embeded-sdk-wifi-ble-bk7231t/platforms/bk7231t/bk7231t_os'
make[1]: Entering directory '/cygdrive/d/a/tuya-iotos-embeded-sdk-wifi-ble-bk7231t/platforms/bk7231t/bk7231t_os'
echo prerequirement0
prerequirement0

Build template_demo

echo prerequirement1
prerequirement1
mkdir -p Debug
echo prerequirement2
prerequirement2
  \033[0;32mCC   beken378/app/app_bk.c\033[0m
../toolchain/gcc-arm-none-eabi-4_9-2015q1/bin/arm-none-eabi-gcc: ../toolchain/gcc-arm-none-eabi-4_9-2015q1/bin/arm-none-eabi-gcc: cannot execute binary file
make[1]: *** [application.mk:785: Debug/./beken378/app/app_bk.o] Error 126
make[1]: Leaving directory '/cygdrive/d/a/tuya-iotos-embeded-sdk-wifi-ble-bk7231t/platforms/bk7231t/bk7231t_os'
make: *** [Makefile:28: mp] Error 2
make: Leaving directory '/cygdrive/d/a/tuya-iotos-embeded-sdk-wifi-ble-bk7231t/platforms/bk7231t/bk7231t_os'

xjikka, 2024-05-29:
> I am using cygwin setup as shown here: https://www.elektroda.com/rtvforum/topic3850712.html

Are You using 64bit Cygwin?

xjikka, 2024-05-29:
I fugure it out, I've downloaded https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q2-update [gcc-arm-none-eabi-4_9-2015q2-20150609-win32.zip](https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q2-update/+download/gcc-arm-none-eabi-4_9-2015q2-20150609-win32.zip) ([md5](https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q2-update/+download/gcc-arm-none-eabi-4_9-2015q2-20150609-win32.zip/+md5))
And replaced compiler in the directory platforms\bk7231t\toolchain\gcc-arm-none-eabi-4_9-2015q1 with the files from zip
Now compile demo works. 
Thank You

xjikka, 2024-05-29:
Finally solved. The problem was on my end ;-)
I've discovered, that the archive OpenBK7231T-master.zip I downloaded earlier is somehow broken (some files are not extracted - e.g. crt0.o)
I re-downloaded OpenBK7231T-master.zip again and now it works OK.

_By the way, this still doesn't work for me: https://github.com/tuya/tuya-iotos-embeded-sdk-wifi-ble-bk7231t (but I don't need it)_

closed 2024-05-29 by xjikka