openshwprojects/OpenBK7231T_App #1248: Configured channel sanity/conflict check logic

divadiow, 2024-05-25:
Not sure what's involved, but I'm a +1 voice for having some kind of channel selection check on the module configuration page to throw an exception or whatever if any of the non-preference channel selections conflict. eg if BAT_Relay and BAT_ADC are on the same channel it'll error on save.

I guess this cannot be done for everything, for example, a user might want 3 LEDs to come on with 1 button toggle, whereas another user might have a device that has multiple LEDs and buttons and they're to be set as pairs. The sanity check would only be able to apply certain components.

Is this a legitimate request or is my thinking/understanding wrong?

MaxineMuster, 2024-05-25:
I really like the idea, it would be a valuable addition to the page.
It might suffer from the possible size of code that might need:
For it's not done with the testing code, there would be the need for some explanation  what's wrong and how to resolve the mismatch.
The advanced users just making a simple mistake would be o.k. with "error, please check configuration", but most users would probably need a bit more text...
But maybe this can be overcome by some kind of automated text derived from the checks (e.g. if the check is (BAT_Relay.channel == BAT_ACD.channel) then text could be simply "on the same channel: BAT_Relay and BAT_ADC", so user can do some research, why this is no valid configuration...)

openshwprojects, 2024-05-25:
Good idea, but the battery driver itself could be also maybe improved.
Maybe it could function without taking channels into the account. I am not
the one who wrote that driver so I am not sure ...