openshwprojects/OpenBK7231T_App #1251: New startup option: do nothing?

jschwalbe, 2024-05-26:
**Describe the bug**
I am using one of the Tuya water valves on my water mains. I almost never want this to be turned off. However, whenever the beken device is rebooted, it defaults to OFF state.  cfg_startup is set to `-1`

**Firmware:**
Build on May 26 2024 13:00:06 version 1.17.594
- Device? Tuya Smart Valve (UJV Ufairy)
- Chip/model: BK7231T
- Device config: cfg_startup -1

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Configure Startup
2. Enter `-1`
3. Ensure that the device is "ON"
4. Restart device
5. Now it is "OFF"

**Screenshots**
unnecessary

**Additional context**
I would love to see an option for cfg_startup which states "do nothing until instructed to change the current state"

Thanks!

DeDaMrAzR, 2024-05-28:
Login to your device, and start webapp, go to config tab and you should have something like this and if possible share that:

[image]

jschwalbe, 2024-06-01:
[image]
and the filesystem doesn't have an autoexec or anything, for that matter.
[image]

my startup settings:
[image]

restarting from the menu works as expected. it remains ON.
unplugging the device and *quicking* plugging it back in, it remains ON.
**HOWEVER**, unplugging the device for 30 secobds and then plugging it back on, it turns OFF. And it turns off *immediately*, before it could even have time to connect to WiFi (I think).
Here is the logfile from a reboot:
```
### Boot Count 111 #######
Warn:CFG:CFG_InitAndLoad: Correct config has been loaded with 15 changes count.
Error:CMD:lfs is absent
Info:GEN:PIN_SetupPins pins have been set up.
Info:MAIN:Main_Init_Before_Delay done
Info:MAIN:Main_Init_Delay
Info:GEN:24 Button_OnInitialPressDown
Info:GEN:24 key_short_press
Info:MQTT:Channel has changed! Publishing 0 to channel 0 
Info:CFG:####### Flash Save Channel 0 as 0 #######
Info:MAIN:Main_Init_Delay done
Info:MAIN:Main_Init_After_Delay
Info:MAIN:Using SSID [xxx]
Info:MAIN:Using Pass [xxx]
Info:MQTT:MQTT_RegisterCallback called for bT water_shutoff_valve/ subT water_shutoff_valve/+/set
Info:MQTT:MQTT_RegisterCallback called for bT bekens/ subT bekens/+/set
Info:MQTT:MQTT_RegisterCallback called for bT cmnd/water_shutoff_valve/ subT cmnd/water_shutoff_valve/+
Info:MQTT:MQTT_RegisterCallback called for bT cmnd/bekens/ subT cmnd/bekens/+
Info:MQTT:MQTT_RegisterCallback called for bT water_shutoff_valve/ subT water_shutoff_valve/+/get
Error:CMD:LFS_ReadFile: lfs is absent
Info:CMD:CMD_StartScript: failed to get file autoexec.bat
Info:MAIN:Main_Init_After_Delay done
Info:MAIN:Time 1, idle 241434/s, free 89304, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 2, idle 241463/s, free 89304, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 3, idle 242470/s, free 89304, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 4, idle 244402/s, free 89304, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 5, idle 244453/s, free 89304, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:ssid:xxx key:xxx
Info:MAIN:Time 6, idle 235869/s, free 83712, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Boot complete time reached (5 seconds)
Info:CFG:####### Set Boot Complete #######
Info:MAIN:Time 7, idle 229149/s, free 83648, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 8, idle 113656/s, free 82456, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9, idle 0/s, free 82456, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTING - 1
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED - 4
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED - 4
Info:MAIN:Time 10, idle 106848/s, free 83408, MQTT 0(0), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:GEN:dhcp=0 ip=192.168.25.79 gate=192.168.25.1 mask=255.255.254.0 mac=10:5a:17:6e:cb:6f
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-55,ssid=xxx,bssid=d6:21:f9:c6:4b:96,channel=6,cipher_type:CCMP
Info:MAIN:Time 11, idle 240023/s, free 83448, MQTT 0(0), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MQTT:mqtt_userName mqtt
mqtt_pass xxx
mqtt_clientID water_shutoff_valve
mqtt_host xxx:1883
Info:MAIN:Time 12, idle 243428/s, free 83360, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MQTT:mqtt_connection_cb: Successfully connected
Info:MQTT:mqtt_subscribed to water_shutoff_valve/+/set
Info:MQTT:mqtt_subscribed to bekens/+/set
Info:MQTT:mqtt_subscribed to cmnd/water_shutoff_valve/+
Info:MQTT:mqtt_subscribed to cmnd/bekens/+
Info:MQTT:mqtt_subscribed to water_shutoff_valve/+/get
Info:MQTT:Publishing val water_shutoff_valve to water_shutoff_valve/host retain=0
Info:MAIN:Time 13, idle 229368/s, free 83104, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MQTT:Publishing val OpenBK7231T 1.17.594 May 26 2024 13:00:15 to water_shutoff_valve/build retain=0
Info:MAIN:Time 14, idle 248330/s, free 83080, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MQTT:Publishing val 10:5a:17:6e:cb:6f to water_shutoff_valve/mac retain=0
Info:MAIN:Time 15, idle 241725/s, free 83112, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MQTT:Publishing val 44.11 to water_shutoff_valve/temp retain=0
Info:MAIN:Time 16, idle 248523/s, free 83120, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38 
```
I appreciate any insight you might have to share on this! Thanks

openshwprojects, 2024-06-02:
Can you disable MQTT totally and retry? Try if it works without MQTT?

jschwalbe, 2024-06-04:
Unfortunately, MQTT was off when I first noticed the problem.

MaxineMuster, 2024-06-04:
Is the part with the button pressed present in every (re-)start?
```
Info:GEN:24 Button_OnInitialPressDown
Info:GEN:24 key_short_press
```
Is the behavior correct, if you temporarily assign the button to another channel?

DeDaMrAzR, 2024-06-05:
Move button and Relay to channel 1, set -1 for channel 1 and report back.

MaxineMuster, 2024-06-05:
That's another possibility, but I would also try to move button to another channel then the relay.
It could also be that button is just configured to the wrong pin, that's a possible reason for the "pressed" button...

DeDaMrAzR, 2024-06-05:
> That's another possibility, but I would also try to move button to another channel then the relay. It could also be that button is just configured to the wrong pin, that's a possible reason for the "pressed" button...

That may be the case but my suspicion is on the WiFi LED. It's a T device, and most of them are in WB2S form factor and P1 is not exposed at all on them. Maybe the confusion was PWM1 marking on the module which is actually a P7 (or just 7) on that list. 

Reference - https://developer.tuya.com/en/docs/iot/wb2s-module-datasheet?id=K9ghecl7kc479

DeDaMrAzR, 2024-06-05:
1 | VBAT Power supply pin (3.3 V), which is connected to the VBAT pin on the internal IC
2 | PWM2 -> P8 pin 
3 | GND Power supply reference ground pin
4 | PWM1 -> P7
5 | 1RX -> P10 
6 | PWM0 -> P6 
7 | 1TX -> P11 
8 | ADC -> P23 
9 | PWM4 -> P24 
10 | CEN  Low-level reset, high-level active (internally pulled high) Docking IC-CEN
11 | PWM5 -> P26

DeDaMrAzR, 2024-06-12:
@jschwalbe 

Did you resolve your issue? Did any of the suggestions made any difference for your problem?

jschwalbe, 2024-06-12:
Greetings! I have been unexpectedly busy with life and haven’t had time to try your solutions. Next week should be slower and I intend to test it out then! Thank you both so much for the ideas. I will certainly report back.

jschwalbe, 2024-06-24:
> Move button and Relay to channel 1, set -1 for channel 1 and report back.
> 
> ![2](https://private-user-images.githubusercontent.com/43572083/336963377-cf05347e-7942-4736-8b37-c20d9390e39e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkxODQxMDAsIm5iZiI6MTcxOTE4MzgwMCwicGF0aCI6Ii80MzU3MjA4My8zMzY5NjMzNzctY2YwNTM0N2UtNzk0Mi00NzM2LThiMzctYzIwZDkzOTBlMzllLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjIzVDIzMDMyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU2MmI1NTZhNWI4MzQ5OWM5ODI3MDY2YTAwMjQzMDM0MmI0MTJmYTM2ZjJjZWFlN2I3YmY0MmE4ZjY3ODljZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Z7jRXWOxh1SeOuH6ergpUzhVsQP3mJ7Ahf3cZXCQ-7E)

Ok, I did the above changes, and set both startup values to -1 (for Channel 0 and Channel 1). 
Now with cold reboot or reboot from UI it does the following:
Old value: off --> new value on
Old value on --> new value on

So, that's still not acting as expected. I'll try the next thing and comment again.

jschwalbe, 2024-06-24:
> 1 | VBAT Power supply pin (3.3 V), which is connected to the VBAT pin on the internal IC 2 | PWM2 -> P8 pin 3 | GND Power supply reference ground pin 4 | PWM1 -> P7 5 | 1RX -> P10 6 | PWM0 -> P6 7 | 1TX -> P11 8 | ADC -> P23 9 | PWM4 -> P24 10 | CEN Low-level reset, high-level active (internally pulled high) Docking IC-CEN 11 | PWM5 -> P26
> 
> EDIT: all this can be also done from the module's index page under config/config module, there the pins are labeled correctly.

Ok, so here's my starting point: (everything else is 0)
```
P1 (RXD2)  WifiLED
P6 (PWM0) Rel 0
P9 (PWM3) LED 0
P24 (PWM4) Btn 0 0
```
And unfortunately, that's as far as I can get. I'm at http://ip/cfg_pins, is that correct? Where should I be making the above changes? Thanks!

MaxineMuster, 2024-06-24:
Would you mind to try my supposal and move the button to another channel than the relay?
Is the button working as expected?

DeDaMrAzR, 2024-06-24:
> Ok, so here's my starting point: (everything else is 0)
>
> ```
> P1 (RXD2)  WifiLED
> P6 (PWM0) Rel 0
> P9 (PWM3) LED 0
> P24 (PWM4) Btn 0 0
> ```
>
> And unfortunately, that's as far as I can get. I'm at [...]

Try and separate the channels for all of your pins, or just separate btn and rel to 1 and 2 for example. If you want your relay to remember the state it was previously in only for that channel set startup option to be -1 the rest should be 0. Try and post a screen shot of your pin config (from http://ip/app?) and your startup (http://IP/cfg_startup).

jschwalbe, 2024-06-24:
> Would you mind to try my supposal and move the button to another channel than the relay? Is the button working as expected when they are on the same channel?
> 
> Since the log seems to indicate the button is pressed on startup, I suspect it to be configured on the wrong pin.

I don't need the button to do anything; however as it currently is set up, the button works as expected. Would it be correct to try your idea but simply removing any reference to the button?
[image]
to this
[image]

?

MaxineMuster, 2024-06-25:
> I don't need the button to do anything; however as it currently is set up, the button works as expected. Would it be correct to try your idea but simply removing any reference to the button?

Yes, that should work to verify or disprove the suspect about the button.

If you have the possobility to check startup logs: is the part with  `Button_OnInitialPressDown` still present?

mindstorm88, 2025-01-21:
Hi guys, my first time posting, i do have the same problem .

Reading all this post , i have just tried MaxineMuster solution, Removing button from 24 in the config. 

Behavior with the startup -1 is working with this solution.

Yes Button press disappear from the startup log. 

But now Hardware button does not work.

mindstorm88, 2025-01-21:
here my latest test, mqtt activated

i did gpio finder, to find that the button  is really at 24 as a dinput. So i put dinput at 24 channel 1 in the config.

hardware button work and startup -1 behavior is good.

funny thing is that i put back btn at 24 instead of dinput. Everything still work properly. 

Problem solved but why ???

mindstorm88, 2025-04-15:
Sadly, not fixed, after a few hours power loss the valve did not stay at previous position !!!