openshwprojects/OpenBK7231T_App #1258: TuyaMCU on UART2 (BK7231N)

DJVirus1980, 2024-06-04:
How to connect the TuyaMCU driver to the UART2 on BK7231N.

NonPIayerCharacter, 2024-06-04:
Enable flag 26 ([UART] Use alternate UART for BL0942, CSE, TuyaMCU, etc).

openshwprojects, 2024-06-04:
Interesting, it's the first time I see that. Is there really a device with TuyaMCU on log port?

DJVirus1980, 2024-06-05:
This termostat
[image]

DJVirus1980, 2024-06-05:
[image]

DJVirus1980, 2024-06-14:
module not receive data from mcu on uart2, if i am conect mcu to uart1 all data received.

Hard-3d, 2024-07-07:
hello. I purchased the same thermostat. how can I make it work with OpenBK?

Hard-3d, 2024-07-07:
[image]

Hard-3d, 2024-07-09:
Please send your own /autoexec.bat

DJVirus1980, 2024-07-09:
startDriver TuyaMCU
startDriver NTP
ntp_timeZoneOfs 3
tuyaMcu_setBaudRate 38400
uartSendHex 55AA000300010306
delay_s 20
tuyaMcu_sendCurTime
uartSendHex 55AA0008000007

LucaArena, 2024-09-13:
Hello; I have a the same issue, tring  to use the UART2: I have an energy power meter with 2 BL0942 chips : one over UART 1 and an other over UART2: I can send data command by UART2 but I cannot receive data: did you solved ?

xjikka, 2024-11-11:
@LucaArena Hi, I have same device energy power meter with 2 BL0942 chips : one over UART 1 and an other over UART 2.
Did You find the solution? On second uart i'm getting just zeroes.

LucaArena, 2024-11-12:
Hi, 

 

Yes I modified the driver to support both chips:

 

Actually I’m testing the solution over 3 devices: it works with good precision (<  1% ) of total energy accumulated:

 

Just a note about software driver implementation:

 

The Interrupt Service Routine over the 2nd UART does not correctly run, so I apply a patch with polling request a 1ms.

 

Cheers

xjikka, 2024-11-12:
@LucaArena Thx, You're using uart_read_byte or test_ty_read_uart_data_to_buffer ?

LucaArena, 2024-11-13:
I’m using  test_ty_read_uart_data_to_buffer: anyway this function calls using uart_read_byte.

 

The power measures are ok:  there is an issue on the total energy: sometime it saved a wrong big value!

xjikka, 2024-11-13:
> I use test_ty_read_uart_data_to_buffer: anyway, this function calls with uart_read_byte. >

I am not able to receive any data from the UART2 using these functions (but the data on the pin is coming if I test it with an external receiver]), 
Can you send me the changed code? I couldn't find it in your repo branches.

LucaArena, 2024-11-13:
Hi, 

 

Yes I had the same issue: the interrupt service for UART2 appends, but the data is not available in the register data input:

 

I changed the strategy, to read the data from UART2: instead of using interrupt, I use a polling: I check if any data is available every 1mS:  is a patch ( not optimized ) 

 

Maybe you cannot find in the branch because is a part of platform ( I cannot remember ) 

 

Anyway, I ‘d like help you: how can I share the code?  By Google Drive ? 

 

I need time to create a good branch, so everyone could use.

 

Cheers

xjikka, 2024-11-13:
>I ‘d like help you: how can I share the code? By Google Drive ?>

Can You send me simply the changed files or code snippets? 
Or You can simpy use any network service like https://www.uschovna.cz/en/?set_lang=en  (i'm using it fo years, no registration, just need email).

xjikka, 2024-11-15:
@LucaArena  I already found the solution, you don't need to send me anything.