openshwprojects/OpenBK7231T_App #1261: BK7231T 1.17.517 OTA not working on one particular device

ingoratsdorf, 2024-06-05:
**Describe the bug**
Had several attempts to upgrade a BK7231T 1.17.517 device via OTA to any other version, tried a few. Not working, not via webapp OTA nor via config->OTA.

**Firmware:**
- Version: 1.17.517
- Device: Arlec Grid Connect PC191 Smart Plug-In Socket with Energy Meter
- Chip/model: BK7231T
- Device config?
```
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Full Device Name Here",
  "model": "enter short model name here",
  "chip": "BK7231T",
  "board": "TODO",
  "flags": "-937425888",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "6": "Rel;1",
    "7": "BL0937CF;1",
    "8": "BL0937CF1;1",
    "10": "WifiLED_n;1",
    "11": "Btn;1",
    "24": "BL0937SEL;1",
    "26": "LED;1"
  },
  "command": "",
  "image": "https://obrazki.elektroda.pl/YOUR_IMAGE.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic_YOUR_TOPIC.html"
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to device web page, open webapp and select RBL file
2. Click OTA
device reboots back into same firmware

**Additional context**
I have quite a few of those plug. this is the only one that cannot be updated.
Funny enough, I did a few updates on this before I ended up with it's current firmware version.

When using webapp OTA, logs:
```
Info:MAIN:Time 236, idle 245304/s, free 42784, MQTT 1(1), bWifi 1, secondsWithNoPing 154, socks 3/38 
Info:OTA:134000
Info:OTA:135000
Info:OTA:136000
Info:OTA:13c000
Info:OTA:13d000
Info:OTA:143000
Info:MAIN:Time 238, idle 269732/s, free 26640, MQTT 1(1), bWifi 1, secondsWithNoPing 156, socks 4/38 
Info:OTA:144000
Info:OTA:145000
Info:OTA:14a000
Info:OTA:14b000
Info:OTA:14c000
Info:MAIN:Time 239, idle 132752/s, free 27056, MQTT 0(1), bWifi 1, secondsWithNoPing 157, socks 4/38 
Info:OTA:14d000
Info:OTA:14e000
Info:OTA:151000
Info:OTA:152000
Info:OTA:153000
Info:OTA:154000
Info:OTA:155000
Info:OTA:156000
Info:OTA:158000
Info:OTA:159000
Info:MAIN:Time 240, idle 171941/s, free 27056, MQTT 0(1), bWifi 1, secondsWithNoPing 158, socks 4/38 
Info:GEN:dhcp=0 ip=192.168.1.26 gate=192.168.1.254 mask=255.255.255.0 mac=38:1f:8d:e7:9b:df
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-56,ssid=Pukeatua,bssid=10:27:f5:85:6d:0e,channel=8,cipher_type:CCMP
Info:OTA:15a000
Info:OTA:15b000
Info:OTA:15c000
Info:OTA:15d000
Info:OTA:15f000
Info:OTA:160000
Info:OTA:161000
Info:OTA:162000
Info:OTA:163000
Info:OTA:164000
Info:OTA:167000
Info:OTA:168000
Info:OTA:169000
Info:OTA:16a000
Info:OTA:16b000
Info:MAIN:Time 241, idle 253010/s, free 27056, MQTT 0(1), bWifi 1, secondsWithNoPing 159, socks 4/38 
Info:OTA:16c000
Info:OTA:176000
Info:OTA:177000
Info:OTA:17d000
Info:MAIN:Time 242, idle 265030/s, free 27056, MQTT 0(1), bWifi 1, secondsWithNoPing 160, socks 4/38 
Info:OTA:17e000
Info:OTA:184000
Info:OTA:185000
Info:OTA:18c000
Info:OTA:18d000
Info:OTA:193000
Info:OTA:194000
Info:OTA:19a000
Info:OTA:19b000
Info:OTA:19c000
Info:OTA:19d000
Info:OTA:19e000
Info:OTA:19f000
Info:OTA:1a0000
Info:OTA:1a1000
Info:OTA:1a2000
Info:MAIN:Time 244, idle 269930/s, free 27056, MQTT 0(1), bWifi 1, secondsWithNoPing 162, socks 4/38 
Info:OTA:1a3000
Info:OTA:1a9000
Info:OTA:1aa000
Info:MAIN:Time 246, idle 244676/s, free 42872, MQTT 0(1), bWifi 1, secondsWithNoPing 164, socks 3/38 
Info:MAIN:Module reboot in 1...
```

When using Config->OTA, logs:
```
Info:HTTP:http_fn_ota_exec: will try to do OTA for https://github.com/openshwprojects/OpenBK7231T_App/releases/download/1.17.603/OpenBK7231T_1.17.603.rbl 
Info:HTTP_CLIENT:Parse url https://github.com/openshwprojects/OpenBK7231T_App/releases/download/1.17.603/OpenBK7231T_1.17.603.rbl
Info:HTTP_CLIENT:host: 'github.com', port: 80
Info:HTTP_CLIENT:HAL_TCP_Establish: created socket 3

Error:HTTP_CLIENT:success to establish tcp, fd=3
Info:OTA:init OTA, startaddr 0x132000
Info:OTA:
myhttpclientcallback state 0 total 0/0
Info:HTTP_CLIENT:httpclient_recv 167 bytes has been read
Error:HTTP_CLIENT:Could not found
Error:HTTP_CLIENT:httpclient_recv_response is error,ret = -43
Info:OTA:
Info:OTA:close OTA, addr 0x132000
Info:OTA:
myhttpclientcallback state 2 total 0/0
Info:OTA:Rebooting in 1 seconds...
```

openshwprojects, 2024-06-05:
hm i think we don't have HTTPS support, and it looks like you're trying to OTA via https?

openshwprojects, 2024-06-05:
if web app ota fails, have you tried to disable all drivers before doing that OTA?

openshwprojects, 2024-06-05:
stopDriver BL0937

ingoratsdorf, 2024-06-05:
Yes, clicked on "Stop all drivers" in webapp tools. Same result.

ingoratsdorf, 2024-06-05:
Weirdest thing is, after I unplugged the device and plugged it back in, it restarted with the latest flashed version. Is there some weird "fast" reboot that reboots the same firmware even though a newer one is installed? I would not think so but have no other ideas :-(
I noticed that the reboots that it did were really fast. Faster than normal. So only conclusion is that it may not have rebooted correctly.

Wolfieeewolf, 2024-08-14:
This has been somewhat solved [here](https://www.elektroda.com/rtvforum/topic3944452.html)

To update you need to OTA first then wait until the device has fully rebooted. Once rebooted remove the device from the socket. Just turning off the power doesn't seem to have success every time.  You then have to hold the button on top, plug it back in and power up the device while holding the button for about 5 -10 seconds. You will know if it works as it will have a long boot-up time and the LED will not light up for about 10 seconds or so. it will then reboot and the LED will start flashing. Wait about 30 seconds to 1 minute for the device to fully boot up before doing anything.  Check the web app and you should see it has hopefully been updated 

I believe doing it this way is like using the CEN pin on the chip and putting it into a higher mode so it can write the firmware. So far has worked every time for me doing it this way.

I will note that these devices are known for blowing the capacitors and will randomly start rebooting themselves. over time. Even if you have power saving on from the start. I'm not sure if it's a firmware fault or not but I have had 5 of these go in the last 12 months both T and N models. 

The N model is more susceptible to blowing the capacitors than the T model.  If you can get your hands on the black-coloured ones they seem to not have this issue so far.

You can replace the capacitors with better ones but it requires opening up the device. I found wrapping it in a thick towel and lightly tapping each side with a hammer works well to loosen the glue. 

I've not replaced any of mine due to potential fire hazards. There is a reason it's blowing the capacitors and it is only doing it with the custom firmware so I don't want to risk it. They are also pretty cheap and it's not worth burning my house down over it..

I have replaced most of mine with Tapo P110's. They seem to be better at power-recording smaller changes and I have not had any issues with them yet. They don't use custom firmware so if you want fully local then you might need to look for something else.

jkwim, 2024-10-17:
I also observed similar issue with 2 x WHDZ03 Smart Plugs recently. The logs indicated that OTA Upgrade was completed but upon reboot they show old firmware version.

Just repeatedly trying made the upgrade successful finally.

Wish that I had seen this thread before.

paoloantinori, 2026-10-31:
I hit the same failure shape on three bulbs on 1.18.219: web UI uploads that looked accepted, the OTA progress counter moved, the version never changed after reboot. Here is what fixed it and what I believe was actually broken.

The fix: upload the .rbl with a raw POST instead of the browser, then a plain reboot.

```
curl -m 600 --data-binary @OpenBK7231T_1.18.320.rbl -H "Content-Type: application/octet-stream" http://DEVICE/api/ota
curl -X POST http://DEVICE/api/reboot
```

The update applied immediately on the soft reboot, on all three bulbs. No socket removal, no safe mode, no button hold.

The diagnostic that explains it: after the failed attempts I read the start of the OTA partition back.

```
curl http://DEVICE/api/flash/132000-100 | xxd
```

The partition still contained a much older raw image from a previous experiment. Those web UI uploads had never written a single byte, not even an erase. The progress counter is not proof of a write: the code increments it when the POST handler starts, before any body data arrives. So "OTA completed but version unchanged" can simply mean the upload never landed.

If you are stuck on this, read the partition head after an upload. First four bytes `RBL` (0x52 0x42 0x4C 0x00) mean the image landed and the next reboot should apply it. Your old firmware's bytes mean nothing wrote, and the upload path itself is the suspect.