openshwprojects/OpenBK7231T_App #1264: Pulse Mode

booroondook-eng, 2024-06-11:
How to implement pulse mode?
After turning on, I need my relay automatically turn off after a while (for example, 5 seconds).
Tasmota has a special command for this mode: "PulseTime<x> <value>"?
But here I did not find anything like this in the configuration and documentation.

DeDaMrAzR, 2024-06-11:
You can use scripting to do that (located under webapp/filesistem/create file and name it autoexec.bat, for example:

delas_s 5
ToggleChannel 1

it will execute on every startup assuming your relay is channel 1 (ToggleChannel <channel number>)

You can also use this as a startup command from your index page, under Config/Change startup command and enter:

addRepeatingEvent 5 1 ToggleChannel 1

It will also execute on every startup where addRepeatingEvenet <time in seconds> <number of repeats> <command to execute>

If you need more info let us know.

booroondook-eng, 2024-06-12:
Perhaps I expressed myself incorrectly and that is why you misunderstood me.
I don't need the relay to switch at startup.
I need the relay to automatically turn off after N seconds after each turn on.

openshwprojects, 2024-06-12:
[image]
?__?
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/autoexecExamples.md

booroondook-eng, 2024-06-12:
Many Thanks!!!