openshwprojects/OpenBK7231T_App #1282: Outdated Home assistant MQTT discovery message format

makserge, 2024-06-29:
**Describe the bug**
This firmware have outdated Home assistant MQTT discovery message format.

**Firmware:**
- Version: 1.17.622
- Device: ONENUO ZHQ01 (Delivered from Aliexpress with product title "ONENUO Tuya Smart Micro USB Adapter Switch 5V WiFi Mini USB Power Adapter Works with Alexa Hey Go Ogle Alice for Smart Home"
- Chip/model: BK7231N
- Device config

{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "USB Smart Adapter HC-S050-WIFI",
  "model": "HC-S050-WIFI",
  "chip": "BK7231N",
  "board": "TODO",
  "flags": "1024",
  "keywords": [
    "5V",
    "MOSFET"
  ],
  "pins": {
    "6": "WifiLED_n;0",
    "7": "Rel;1",
    "8": "Btn_Tgl_All;0",
    "24": "Btn;1"
  },
  "command": "",
  "image": "https://obrazki.elektroda.pl/5839545100_1692292748_thumb.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/viewtopic.php?p=20697534#20697534"
}

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'http://device_ip'
2. Click on 'Config'
3. Click on 'Home Assistant Configuration'
4. Press 'Start Home Assistant Discovery Configure MQTT'
5. Go to to HA web interface and check for new just added device
6. Actual result: device not added 

Reason: firmware has outdated MQTT discovery format 
Correct format 
according to HA docs
(https://www.home-assistant.io/integrations/mqtt/#mqtt-discovery)

Example

topic 
homeassistant/switch/OpenBK7231N_LedStripHall_relay_0/config

payload

{
    "device_class":"switch",
    "device": {
        "identifiers": "OpenBK7231N_LedStripHall_relay_0",
        "manufacturer": "Led Strip Hall",
        "model": "Led Strip Hall",
        "name": "Led Strip Hall"
    },
    "object_id":"OpenBK7231N_LedStripHall_relay_0",
    "unique_id": "OpenBK7231N_LedStripHall_relay_0",
    "name": "Led Strip Hall",
    "command_topic": "ed_strip_hall/0/set",
    "payload_on": "1",
    "payload_off": "0",
    "state_topic": "led_strip_hall/0/get",
    "state_on": 1,
    "state_off": 0,
    "qos": 1
}

rpv-tomsk, 2024-06-30:
Current discovery message format is JSON in all cases (in application code).
So, your provided description looks unreal for me. But I'm newbie here so I could be wrong.

openshwprojects, 2024-06-30:
Aren't you confusing the old yaml code displayed for legacy purposes with the internal HASS discovery format?

makserge, 2024-06-30:
> Current discovery message format is JSON in all cases (in application code). I had no issues when did sensors discovery from my device to my hass. So, your provided description looks unreal for me. But I'm newbie here so I could be wrong.

What HASS version do you use?

rpv-tomsk, 2024-06-30:
2024.6.4

closed 2024-07-13 by makserge