openshwprojects/OpenBK7231T_App #1284: What are the reasons for frequent device restarts?

q127981, 2024-07-01:
I have a BL602 device.
After configuring wifi and mqtt information for the device, the device works normally.
Then disconnect the WiFi Internet (keep the device connected to WiFi), and the device will restart frequently and automatically.
The reason for displaying the error is: BL-RST-SOFTWARE-WATCHDOG
But I have disabled the watchdog, it should not work.
I want to ask how to prevent the device from being restarted frequently when the connected WiFi has no Internet?

The log of the device is always lost after reboot. I can't get the reason why the device restarts frequently from the log.

**Firmware:**
- Version  1.17.628
- Device BL602
- Chip/model: BK7231T?
- Device config
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Full Device Name Here",
  "model": "enter short model name here",
  "chip": "BL602",
  "board": "TODO",
  "flags": "1024",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "14": "wifiLED_n;1",
    "20": "Btn;1",
    "21": "Rel;1"
  },
  "command": "",
  "image": "https://obrazki.elektroda.pl/YOUR_IMAGE.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic_YOUR_TOPIC.html"
}

rpv-tomsk, 2024-07-01:
Catch the same BL-RST-SOFTWARE-WATCHDOG on BL602.

giedriuslt, 2024-07-02:
You cannot disable software watchdog, it always works; 
what is our mqtt config? does it contain dns name? could you try replacing it with IP?

q127981, 2024-07-03:
> You cannot disable software watchdog, it always works; what is our mqtt config? does it contain dns name? could you try replacing it with IP?

Yes, I disabled the watchdog and it still works.
My mqtt config:
`host mqtt.xxxxx.com
port 1883
username  xxxx
password xxxx
`
I changed host to ip, and this problem never occurred again. thank you very much!
@giedriuslt

giedriuslt, 2024-07-03:
@openshwprojects maybe we can make dns resolution for mqtt asynchronous? Now it's in the main loop, and if slow (or no-internet) causes reboot

restoration2699, 2024-07-22:
> I changed host to ip, and this problem never occurred again. thank you very much!

did almost the same, except mqtt server in local network.
but still get "BL_RST_SOFTWARE_WATCHDOG".
how I can permanently disable watchdog function?

openshwprojects, 2024-07-23:
If you disable watchdog, then in case of crash, device will stay offline until you manually power it off and on again..

openshwprojects, 2024-07-23:
@giedriuslt  maybe move it to threaD?

giedriuslt, 2024-07-27:
Made a fix for this here #1298 
No need to use threads lwip can resolve dns asynchronously

PR #1298 (2024-08-12): Resolve MQTT DNS asynchronously
This makes use of asynchronous dns resolution in lwip, fixes issues with restarts if internet breaks. tested on bk7231n and bl602.
Fixes #1284. 
XR809 uses very old lwip so there is no way to ask for ipv4 ip, but it probably does not support ipv6 anyway, so should be no issue
closed 2024-08-12 by openshwprojects