openshwprojects/OpenBK7231T_App #1292: OTA is broken for BL602

giedriuslt, 2024-07-19:
**Describe the bug**
OTA is broken 

**Firmware:**
- Version [e.g. 1.17.638]
- Chip/model: BL602
- Device config?

**To Reproduce**
Steps to reproduce the behavior:
Update fails, device crashes, log shows this:
```
[19:51:58.437] - Info:MQTT:Publishing val 3 to obl4C96B69B/sockets retain=0
Info:MAIN:Time 271, idle 0/s, free 103456, MQTT 1(1), bWifi 1, secondsWithNoPing 200, socks 3/21
[19:51:59.432] - Info:MQTT:Publishing val -63 to obl4C96B69B/rssi retain=0
Info:MAIN:Time 272, idle 0/s, free 103456, MQTT 1(1), bWifi 1, secondsWithNoPing 201, socks 3/21
[19:52:00.353] - Current task sp data:
RA:2308b7f8, mstatus:80007880
A0:4200e408 A1:4202d018 A2:4200e99c A3:4200e99c A4:00000808 A5:4202be80 A6:4202be80 A7:4202324c
T0:23030290 T1:0000000f T2:a5a5a5a5 T3:a5a5a5a5 T4:a5a5a5a5 T5:a5a5a5a5 T6:a5a5a5a5
S0:00000808 S1:80000408 S2:4202d010 S3:00000800 S4:c202d418 S5:4200e408 S6:4202d018 S7:a5a5a5a5
S8:a5a5a5a5 S9:a5a5a5a5 S10:a5a5a5a5 S11:a5a5a5a5
Exception Entry--->>>
mcause 38000005, mepc 2308b834, mtval c202d41c
Exception code: 5
  msg: Load access fault
[19:52:02.794] - Starting bl602 now....
Booting BL602 Chip...
```

Maybe realloc causing trouble?

giedriuslt, 2024-07-19:
Works with fine  637

giedriuslt, 2024-07-19:
Luckily recovery IS possible. I set up network throttling to 8kb/s like this and it worked, old firmware was uploaded, took 7 minutes.
[image]

openshwprojects, 2024-07-20:
That's strange, I reverted for BL602:
https://github.com/openshwprojects/OpenBK7231T_App/commit/c00f1a968de368ecf64f22114d1b152e96165685
Maybe realloc is not working on BL602?

Why do you think that the code is incorrect? recv will return 0 or -1 if there is no more data.

The folowing change (realloc) was added so we can receive HTTP packets longer than 1024 bytes.

giedriuslt, 2024-07-20:
Yes, so that is the issue, ota data size is 400KB, it tries to allocate all of that, and then there is no memory left to do anything else.
This code should have some limit. Not exactly sure why this even works for Beken.

openshwprojects, 2024-07-20:
I didn't expect that, it passed all tests on Beken. So, shall we introduce a limit? I don't know, 8kB?

giedriuslt, 2024-07-20:
I don't know what is required, but I would say having limit would be good here, like 8kb or 16kb or 32kb...

giedriuslt, 2024-07-20:
1.17.647 OTA again works good for BL602