openshwprojects/OpenBK7231T_App #1304: Multi Device Issues

jonny190, 2024-08-08:
**Describe the bug**
When i have more then one LN882H device connected to my Unifi WiFi only the first connected one of them works, the others go in to a reboot cycle, despite being shown as connected in the Unifi Controller

**Firmware:**
- Version: 1.17.652
- Device: UK 3 Pin Relay (BSD48)
- Chip/model: LN882H
- Device config

`
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Smart Socket",
  "model": "BSD48",
  "chip": "LN882H",
  "board": "TODO",
  "flags": "1024",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "0": "LED_n;1",
    "3": "Btn;0",
    "7": "BL0937CF1;0",
    "10": "LED_n;0",
    "11": "Rel;0",
    "12": "BL0937CF;0",
    "19": "BL0937SEL;0"
  },
  "command": "",
  "image": "",
  "wiki": ""
}
`

**To Reproduce**
Steps to reproduce the behavior:
1. Connect first device to power
2. Test web interface 
3. Connect second/third device 
4. Second and third devices enter a reboot cycle and no web interface on second and third devices but all show as connected in Unifi

**Screenshots**
[image]

**Additional context**
Ive Set static IP's on 2 devices and even changed the unique MAC addresses to try and solve this

openshwprojects, 2024-08-08:
Hey, just to be sure, what are the MACs of the devices as seen by router? Are they the same?

jonny190, 2024-08-08:
Afternoon, yeah these are the same Mac addresses on the AP's as shown on the router and on the device config pages

MaxineMuster, 2024-08-11:
If the MAC addresses are equal, are they different from the default defined in `sdk/OpenLN882H/components/utils/sysparam_factory_setting.h` (00:50:C2:5E:10:88)? The startup code should generate a (different) random MAC in that case, if not, we should investigate why.

If they are equal but different OpenLN882h will accept them. Then maybe they were equal even before you installed the new firmware?
How do you know the devices are in a reboot cycle? Did you check the serial logs on TX2? 
(Never do this, if connected to mains, GND will probably be connected to live power! 
Only if LN882H powered by a isolated 5V/3.3V power supply.)

If you connect only one device, you can access and change it?
Static IP won't work on LN882H, but setting the MAC should work (note, the device name is not changed automatically if you set the MAC).

If the MAC addresses are different, I don't see why it shouldn't work.

giedriuslt, 2024-08-13:
I do get the hostname based on default mac (because host name seems to be generated before mac is changes), so if you connect multiple ones you probably will only get to see single one, and it might appear as reboots maybe. Not sure how ruoters behave with such

jonny190, 2024-08-13:
@MaxineMuster Only reason i have to think they are rebooting is they have relays in and i can hear them ticking on and off just like when they are powering on or power cycled 

Thats how i set the static MAC addresses, powered one on at a time and set each with their own randomly genorated MAC address (not that i'm saring the mac address is the issue, just that would be my first guess coming from a networking background) 

I'm not sure if its a unifi issue or firmware, but ive had no issues with ESP8266 or ESP32 devices

MaxineMuster, 2024-08-14:
I see, thanks .
Best way to investigate would in fact be to see the log on Tx2, but that's some trouble including soldering and another power supply...

If they don't power on completely, they will fall back to safe mode an be only available with their own AP. But they are still visible on the APs, right?

If reboot takes place later, e.g. from watchdog, can you reach them via ping during that period after startup before it restarts?

Did you configure anything else, especially MQQT, with a hostname?
That might be an issue which had been addressed just with the latest release.

divadiow, 2024-08-20:
> (00:50:C2:5E:10:88)? The startup code should generate a (different) random MAC in that case, if not, we should investigate why.

Sadly it does not. I see this mac a lot. I do not have any LN devices in actual use but can fire up some for simultaneous testing if need be.

divadiow, 2024-08-20:
Today: https://www.elektroda.com/rtvforum/viewtopic.php?p=21197784#21197784

giedriuslt, 2024-08-20:
I do not see any mac in the threads (except some from original device which is completely different). Hostname is created from default mac and will be identical, but mac itself probably is fine. hostname and mac are not the same thing

divadiow, 2024-08-20:
ok sure. I've fired up a few different LNs and they're all diff

[image]

I'd not noticed the mac change between initial AP and then as STA 😳
AP:
[image]
STA:
[image]

second device flashed from non-OBK:
AP
[image]
STA
[image]

giedriuslt, 2024-08-20:
Yes, this is what happens, on initial boot hostaname is generated (while still using default mac), creating ap does not change mac (it's irrelevant aanyway). Connecting to wifi finally changes a mac address. 
This probably could be improved to make it less messy. But not sure how all this is related to reboots if at all.