openshwprojects/OpenBK7231T_App #1305: LittleFS config getting erased when updating from 1.17.251 to 1.17.652

rickgartner, 2024-08-08:
**Describe the bug**
When performing an OTA update LittleFS is being wiped out.  Have to manually recreate atuoexec.bat.

**Firmware:**
- Version [e.g. 1.17.652]
- Device feit light switch
- Chip/model: BK7231T?
-Upgrading from 1.17.251 via OTA tab in web application 

**To Reproduce**
Steps to reproduce the behavior:
1. Launch web application 
2. Select OTA tab
3. Select OTA file from disk
4. Start OTA
5. Once system reboots everything in LittleFS is missing.

divadiow, 2024-08-08:
in an attempt to recreate the scenario you describe I have flashed a WB3S (BK7231T) back to a factory fw from 0x11000 then to OpenBeken 1.17.251, then with a basic simple autoexec present, OTA to 1.17.652.

And yes, LFS is clear after the flash. No autoexec.

Wolfieeewolf, 2024-08-14:
This has been happening for a while 

https://github.com/openshwprojects/OpenBK7231T_App/issues/1010