openshwprojects/OpenBK7231T_App #1309: What's the non-obsolete method for flashing BK7231N?

jfly, 2024-08-11:
I see [Flashing for BK7231N (obsolete method)](https://github.com/openshwprojects/OpenBK7231T_App/blob/6e0dfef4449fd435dd5ac1fbcd468f8680ba4e5a/FLASHING.md#flashing-for-bk7231n-obsolete-method). The title implies that there's a non-obsolete method, but I can't find it documented. Some help, please?

Thanks for the very awesome project!

openshwprojects, 2024-08-11:
https://github.com/openshwprojects/BK7231GUIFlashTool

openshwprojects, 2024-08-11:
[image]

jfly, 2024-08-11:
Ah, sorry I skipped over that part because I'm not on Windows!

What about for Linux?

madleech, 2024-08-16:
For linux use serial flashing; it still works fine even if it's deprecated.