openshwprojects/OpenBK7231T_App #1325: RGB LED Strip flickering during effects (with DDP)

h3ndrik, 2024-08-20:
While displaying color changing effects, the RGB LEDs flicker constantly while the colors change.

Since we have no effects (yet) on device, I'm sending some color changes and color wheels via ledfx or xLights over the DDP protocol. That works as intended. But there is a very annoying flicker going on constantly.

- Firmware Version 1.17.663
- Device: [LSC LED Strip 5m 3203632](https://www.elektroda.com/rtvforum/topic4034357.html)
- Chip/model: BK7231N

Steps to reproduce the behavior:
1. Enable the SM16703P driver
2. Enable the DDP driver
3. Send some dynamic effects with xLights or ledfx (like a color wheel)
4. See error, the RGB LEDs flicker while displaying the effect

openshwprojects, 2024-08-21:
thx for the fix, I will just point out that "Since we have no effects (yet) on device," is not actually true, we have PixelAnim driver:
https://www.elektroda.com/rtvforum/viewtopic.php?p=21101474#21101474

Wolfieeewolf, 2024-08-23:
I had the same issue with WLED. 

You need to not start the LED driver and only start DDP. 

Doing that worked for me. I think there is a conflict where WLED is trying to change the LEDs, and the OBK driver is also doing the same thing, and it gets confused, which causes the flicker.