openshwprojects/OpenBK7231T_App #1338: Not connecting to WiFI

Ienien77, 2024-09-03:
**Describe the bug**
A clear and concise description of what the bug is.

After successfully flashing my device with what is a very old firmware (about 1year old [ i realised that after i had my device not working when looking at recent releases]), i tried to configure WiFi.

Input correct SSID and Password, then device rebooted and nothing happened : i can't have it on to connect to my WiFi.

I really hope i will be able to recover it without having to open it as it's glued ...

**Firmware:**
- Version [e.g. 1.14.129] : 1.17.262 (a bit old apparently if we look at relases)
- Device? ELIVCO  LSPA9 WiFi Smart Plug
- Chip/model: BK7231T
- Device config? Only WiFi setting modified since flashing.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash firmware into it
2. Click on Config
3. Scroll down to Configure WiFi
4. Enter WiFi parameters, let it reboot
5. Nothing happend, doesn't connect to WiFi nor have it's own WiFi.

**Additional context**
Got it flashed with https://github.com/tuya-cloudcutter/tuya-cloudcutter/

DeDaMrAzR, 2024-09-04:
There were a lot of changes and updates and improvements since 1.14.129, one of which is unfortunately breaking config and LFS on T modules especially as more and more devices and driver became supported. 

Solution would be to place your device into a safe mode (5 reboots for example) it should appear in self AP mode and reconfigure your device again.

Ienien77, 2024-09-06:
Oh ! I will probably try that when after i finish soldering a new wire for 2TX for which i broke the solder pad trying to have a UART log console. I also try to connect to uart console but with no success. I had 4 wire connected for it : 
- 1 on VBat pad for 3.3v
- 1 on GND
- 1 on 1RX 
- 1 on 1TX
Tryied it with 115200 baud but did not output anything.

Ienien77, 2024-09-06:
What a cool feature that is safe mode. Make me realise that i've of course made a typo in my SSID name ... 
Thanks for your help, i close that.

closed 2024-09-06 by Ienien77