openshwprojects/OpenBK7231T_App #1348: BK7231N constant crash after "Error:MAIN:Low heap warning!"

OoSTARTERoO, 2024-09-13:
**Describe the bug**
A clear and concise description of what the bug is.

**Firmware:**
- Version 1.17.692
- Device LSPA9
- Chip/model: BK7231N
- Device config 
startup command `backlog startDriver BL0937; PowerSave 1; loglevel 6; ` is set

Logs

`Info:MAIN:Time 12, idle 43017/s, free 63704, MQTT 0(0), bWifi 1, secondsWithNoPing -1, socks 3/38 POWERSAVE
Info:MAIN:Time 13, idle 73866/s, free 52048, MQTT 0(0), bWifi 1, secondsWithNoPing -1, socks 4/38 POWERSAVE
Info:MQTT:mqtt_host empty, not starting mqtt
Info:MAIN:Time 14, idle 75616/s, free 40352, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 5/38 POWERSAVE
Error:MAIN:Low heap warning!
Info:MAIN:Time 15, idle 67394/s, free 16960, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 7/38 POWERSAVE
Debug:HTTP:TCP Client 192.168.0.13:43305 thread creation failed! fd: 8
Error:MAIN:Low heap warning!
Info:MAIN:Time 16, idle 71537/s, free 5264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 8/38 POWERSAVE
Debug:HTTP:TCP Client 192.168.0.13:34089 thread creation failed! fd: 8
Debug:HTTP:TCP Client 192.168.0.13:34601 thread creation failed! fd: 8
Error:MAIN:Low heap warning!
Info:MAIN:Time 17, idle 68771/s, free 5264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 8/38 POWERSAVE
Debug:HTTP:TCP Client (PC's ip):35113 thread creation failed! fd: 8
Debug:HTTP:TCP Client (PC's ip):35625 thread creation failed! fd: 8
Debug:HTTP:TCP Client (PC's ip):49449 thread creation failed! fd: 8
Debug:HTTP:TCP Client (PC's ip):49705 thread creation failed! fd: 8
Debug:HTTP:TCP Client (PC's ip):49961 thread creation failed! fd: 8
Debug:HTTP:TCP Client (PC's ip):36137 thread creation failed! fd: 8
Error:MAIN:Low heap warning!
Info:MAIN:Time 18, idle 49883/s, free 5264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 8/38 POWERSAVE
Error:MAIN:Low heap warning!
Info:MAIN:Time 19, idle 59020/s, free 5264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 8/38 POWERSAVE
bk_reboot
wdt reboot`

`Info:MAIN:Time 12, idle 44069/s, free 63680, MQTT 0(0), bWifi 1, secondsWithNoPing -1, socks 3/38 POWERSAVE
Error:HTTP:TCP Client is disconnected, fd: 2
Info:MAIN:Time 13, idle 67354/s, free 63712, MQTT 0(0), bWifi 1, secondsWithNoPing -1, socks 3/38 POWERSAVE
Info:MQTT:mqtt_host empty, not starting mqtt
Info:MAIN:Time 14, idle 68115/s, free 63728, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 3/38 POWERSAVE
Info:MAIN:Time 15, idle 73756/s, free 28624, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 6/38 POWERSAVE
Error:MAIN:Low heap warning!
Info:MAIN:Time 16, idle 75687/s, free 16928, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 7/38 POWERSAVE
Error:HTTP:TCP Client is disconnected, fd: 7
Error:MAIN:Low heap warning!
Info:MAIN:Time 17, idle 71973/s, free 5232, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 8/38 POWERSAVE
Debug:HTTP:TCP Client 192.168.0.13:28985 thread creation failed! fd: 8
Error:MAIN:Low heap warning!
Info:MAIN:Time 18, idle 63971/s, free 5232, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 8/38 POWERSAVE
Error:MAIN:Low heap warning!
Info:MAIN:Time 19, idle 66038/s, free 5232, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 8/38 POWERSAVE
Error:MAIN:Low heap warning!
Info:MAIN:Time 20, idle 69648/s, free 5024, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 8/38 POWERSAVE
bk_reboot
wdt reboot
`

openshwprojects, 2024-09-14:
Does it happen without loglevel setting?

OoSTARTERoO, 2024-09-14:
Yes, without any startup command

MaxineMuster, 2024-09-14:
Are there any other open browser windows open?
Something is eating your memory very fast (see free memory decreaseing from second to second):
52k
40k
16k
5k

63k
28k
17k
5k

openshwprojects, 2024-09-16:
http server again, per socket malloc....

OoSTARTERoO, 2024-09-16:
Oh... Is there any way that I can help you to solve this issue?
I can dig up deeper If you can't replicate it.

Also, Is there any versions without this issue?