openshwprojects/OpenBK7231T_App #1355: Driver / pin function list is not alphabetic

divadiow, 2024-09-21:
Just a little niggle. 

If anyone has the know-how or interest, maybe the list could always be made alphabetic.

[image]

divadiow, 2024-09-21:
oh. looks like im having a go https://github.com/openshwprojects/OpenBK7231T_App/pull/1356

MaxineMuster, 2024-09-25:
Problem is, that the "list" corresponds to an enum in source code, which is used for the pins "role".
This makes it impossible (or at least very challenging) to change the order in the C code (which would be the simplest approach).

So I would sugest to realize this in JS on the web client.
Made a change in a saved webpage from Windows Simulator.
It's not so much to do, will try to make a PR later, but you might try the two versions in a browser "standalone".

```
--- Configpage_original_w_linebreaks2.html	2024-09-25 11:03:24.168495000 +0200
+++ Configpage_sorted_w_linebreaks2.html	2024-09-25 11:02:55.104623000 +0200
@@ -4,6 +4,9 @@
 
 
 
+ // build a case insensitive (with "sort(Intl.Collator().compare)") array of IO-Roles and their index 
+ // so we can build the options "selects" in a sorted order
+var  sr = r.map((e,i)=>{return e[0]+"#"+i}).sort(Intl.Collator().compare).map(e=>e.split("#"));
 function hide_show() {
 	n=this.name;
 	er=getElement('r'+n);
@@ -23,12 +26,12 @@
 	let s = document.createElement("select");
 	s.className = "hele";
 	s.name = id;d.appendChild(s);
-	for (var i = 0; i < r.length; i++) {
-		if(b && r[i][0].startsWith("PWM")) continue; 
+	for (var i = 0; i < sr.length; i++) {
+		if(b && sr[i][0].startsWith("PWM")) continue; 
 		var o = document.createElement("option");	
 		o.text = r[i][0];	
-		o.value = i;	
-		if (i == c) {
+		o.value = r[i][1];	
+		if (r[i][1] == c) {
 			o.selected = true;
 		}
 		s.add(o);

```

Since I can't upload HTML files, you need to rename the txt files ...

[Configpage_sorted_w_linebreaks.html.txt](https://github.com/user-attachments/files/17129766/Configpage_sorted_w_linebreaks.html.txt)
[Configpage_original_w_linebreaks.html.txt](https://github.com/user-attachments/files/17129767/Configpage_original_w_linebreaks.html.txt)

MaxineMuster, 2024-09-26:
Please take a look to [#PR1361](https://github.com/openshwprojects/OpenBK7231T_App/pull/1361)

closed 2024-10-07 by divadiow