openshwprojects/OpenBK7231T_App #1357: No AHT2X io roles

xpawelsky, 2024-09-21:
**Describe the bug**
No AHT2X IO roles. They are non existent in src. 
No temperature  / humidity reading. 

**Firmware:**
- Version: 1.17.700 with AHT2X driver enabled
- Device: TH01Y-CB3S
- Chip/model: BK7231N
- Device config:

`startDriver TuyaMCU
startDriver tmSensor
startDriver AHT2X
startDriver I2C

// may be needed, depends on device, some also use 9600
tuyaMCU_setBaudRate 115200`

**To Reproduce**
Steps to reproduce the behavior:
1. Compile repo with AHT2X driver enabled
2. Go to 'Config'
3. Click on 'Pin Settings' dropdown, any pin
4. No AHT2X_SDA / AHT2X_SCK IO roles

openshwprojects, 2024-09-22:
I think that you can set pins for this driver just by putting their indexes after startDriver <DriverName> command:
[image]

divadiow, 2024-09-24:
this device is TuyaMCU though? like the one pictured here https://www.elektroda.com/rtvforum/topic4054857.html

divadiow, 2025-01-03:
in the case of builds where including the AHT2X driver does not lead to a drop-down entry in the module config page (eg BK-T https://github.com/openshwprojects/OpenBK7231T_App/pull/1490), the driver can be started as:

**startDriver AHT2X [SCKpin] [SDApin] [TempCH] [HumCH]** eg **startDriver AHT2X 26 24 1 2**