openshwprojects/OpenBK7231T_App #1371: [QUESTION] Making the MQTT Color topic accept the R, G, B format

hegeda14, 2024-09-30:
Hi :)

Tried to dig through the code/issues/readmes, but only found scarce mentions about this. Is there an option to make the firmware accept the 0-255, 0-255, 0-255 format for the LED Colors? All my ESP32 Tasmota devices are using this, OpenHAB is configured by default for this. Thank you!

openshwprojects, 2024-09-30:
Hey, what kind of syntax do you want exactly, do you want:
Color 255 255 255
or
Color 255,255,255
or something else?

openshwprojects, 2024-10-01:
i see, can you try now?
https://github.com/openshwprojects/OpenBK7231T_App/commit/07839162e000ffa529f457db1c255bf8005d7384

hegeda14, 2024-10-02:
Hi :) 
Works wonderfully, thank you for your quick help! Have a splendid day!

closed 2024-10-02 by hegeda14